gereic/GXAirCom #3: PSRAM issue on TTGO Lora32 (T3_v1.6)

bsiege, 2020-09-20:
When flashing 1.4.4 the device stays in a boot-loop. As i understand the log, it has to do with initializing the non-existent psram 

````
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
[W][esp32-hal-psram.c:30] psramInit(): PSRAM init failed!
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401303b0: 00000000 00000040 00004000
Core 0 register dump:
PC      : 0x401303b4  PS      : 0x00060030  A0      : 0x800fc28b  A1      : 0x3ffbbc80
A2      : 0x8001f880  A3      : 0x8001f880  A4      : 0x4008897c  A5      : 0x00000001
A6      : 0x00060c20  A7      : 0x00000000  A8      : 0x800fb458  A9      : 0x3ffbbc60
A10     : 0x00000020  A11     : 0x00000001  A12     : 0x3ffcaf72  A13     : 0x00000001
A14     : 0xffffffff  A15     : 0x00000000  SAR     : 0x0000000f  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe

Backtrace: 0x401303b4:0x3ffbbc80 0x400fc288:0x3ffbbca0 0x400ff7a7:0x3ffbbcd0 0x40091bb1:0x3ffbbcf0
````

Device is ok. SoftRF works.

gereic, 2020-09-21:
Can you send a picture from the board ?
Also a pinout would be nice.

bsiege, 2020-09-21:
Natürlich. Sorry:  
[LILYGO Complete Specs](http://www.lilygo.cn/prod_view.aspx?TypeId=50003&Id=1130&FId=t3:50003:3)

gereic, 2020-09-21:
Just uploaded a new Version, which checks, if device has PSRAM.
Can't check at the moment if it works.
Can you try it ?

sbishop61, 2020-09-21:
Just tried it on a T-Beam:
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
```

bsiege, 2020-09-21:
Had partly success with 1.4.5 but i also realized here was a build_flag on platformio.ini "-D BOARD_HAS_PSRAM" which i now commented out (idiot inside). According to specs i used board=ttgo-lora32-v1 because it belongs to the v1 category.
Now testing as a airwhere groundstation replacment. Looks promising. 
Only actual flaw: Board set as T-Beam is running fine, but not as HELTEC/TTGO LORA, there is also some kind of boot loop.

```
[I][main.cpp:1120] taskBluetooth(): stop task
[I][main.cpp:707] setupWifi(): Setting soft-AP ... 
[I][main.cpp:710] setupWifi(): Ready
[I][main.cpp:715] setupWifi(): Setting soft-AP configuration ... 
[I][main.cpp:717] setupWifi(): Ready
[I][main.cpp:723] setupWifi(): hostname=GXAirCom-07EB3C
[I][main.cpp:763] setupWifi(): my APIP=192.168.4.1
[I][main.cpp:2105] taskStandard(): stop task
[I][main.cpp:2164] powerOff(): wait until all tasks are stopped
[I][main.cpp:2170] powerOff(): switch power-supply off
```

Congratulations so far and thanks for the effort. Coming from airwhere over SoftRF to GXAirCom, this seems to be the most evolved and mature implementation for FANET so far.

gereic, 2020-09-21:
When you select the HELTEC/TTGO LORA board, you have to put pin 34 to VCC because this is the voltage-pin for battery.
Or you can put in a voltage-divider like in the picture, to measure the battery-voltage.
[image]

gereic, 2020-09-21:
On my TTGO Lora-Boards, it works also with the flag in the compiler set.
Can you try, if it works also on your board ??
You can use the files from the bin-directory directly.
The only thing you need is the esp32 download-tool.
Can you try it and let me know if it works ??
[image]

bsiege, 2020-09-21:
Confirm 34 wired to VCC (3.3) and it boots as HELTEC/TTGO-Lora but without OLED. 
If i erase the flash in platform.io with `pio run -t erase` i have to add spiffs.bin anyway with the flash tool.

And finally.. yes it runs now directly flashed too...  Was a bit fiddling between platform.io on macos and esptool on win..

gereic, 2020-09-22:
Updated new Version for direct support of T3_V1.6.
You can directly update firmware.bin and spiffs.bin via webinterface.
Then select the "TTGO LORA T3_V1.6 (V2 1.6)" from the settings.
OLED should also work then.
Please test and let me know if it works.

bsiege, 2020-09-22:
Sorry, no luck. First failed OTA, then tried esp flash tool with new files from bin.

```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
[W][esp32-hal-psram.c:30] psramInit(): PSRAM init failed!
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401305a0: 77005c96 950300c0 285810d0
Core 0 register dump:
PC      : 0x401305a6  PS      : 0x00060030  A0      : 0x00000050  A1      : 0x3ffbbc60
A2      : 0x3f40da93  A3      : 0x00000001  A4      : 0x3ffba3d8  A5      : 0x00000001
A6      : 0xffffffff  A7      : 0x00000000  A8      : 0x800fb740  A9      : 0x3ffbbc40
A10     : 0x00000011  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x3ffc0268
A14     : 0x3ffcafb4  A15     : 0x00000000  SAR     : 0x0000000f  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe

Backtrace: 0x401305a6:0x3ffbbc60 0x0000004d:0x3ffbbc80 0x400fc45c:0x3ffbbca0 0x400ff97b:0x3ffbbcd0 0x40091c2d:0x3ffbbcf0

Rebooting...
```
My only way to success is to compile it with my parameters. And now i know how to prepare/upload spiffs with
 **platformio run --target uploadfs** the platform.io setup works for me without the need for esp tool.

platformio.ini:

```
[env:esp32dev]
platform = espressif32
board = ttgo-lora32-v1
board_build.partitions = min_spiffs.csv
framework = arduino
monitor_port = /dev/tty.usbserial-01BBXXXX
monitor_speed = 115200
upload_port = /dev/tty.usbserial-01BBXXXX
upload_speed = 921600
build_flags = -DCORE_DEBUG_LEVEL=3
extra_scripts = post:extra_script.py
monitor_flags = 
   -f
   esp32_exception_decoder
```

bsiege, 2020-09-26:
> When you select the HELTEC/TTGO LORA board, you have to put pin 34 to VCC because this is the voltage-pin for battery.
> Or you can put in a voltage-divider like in the picture, to measure the battery-voltage.

Found schematics, this Board has already a 100k/100k divider on pin 35. I try to calculate the correct values, but they are yet too low an the Board shuts immediately down on battery power. 

```
#define HELTEC_BAT_PIN 35
vBatt = (100000.0f + 100000.0f) / 100000.0f * (float(adc_reading) / 4095.0f *3.3) ;
```
[Schematics](https://raw.githubusercontent.com/LilyGO/TTGO-LORA32/master/schematic1in6.pdf) (Power supply on latest sheet south/east)

gereic, 2020-09-26:
OK, if the board has already a BAT-PIN on pin 35, then it is equal to the t-beam V07.
I will create a new Version for that and you can try it.

gereic, 2020-09-26:
ok just uploaded new Version.
Can you try it ?
You have to select "TTGO LORA T3_V1.6 (V2 1.6)" in the settings.
The Pin for battery is now IO35 (same as on T-Beam V07)

bsiege, 2020-09-26:
Had a look on schematics of T-Beam V07. It as indeed the same 100k/100k divider.
My actual workaround was:
vBatt = (100000.0f + 100000.0f) / 100000.0f * (float(adc_reading) / 4095.0f ***3.73**) ;
But  i see the T-Beam calculation has another formula which should be more accurate

gereic, 2020-09-26:
give the new version a chance.
And let me know the result.

bsiege, 2020-09-26:
Sorry. Upload via webinterface. Still not booting with 
`psramInit(): PSRAM init failed.`
If i compile with my parameters it works fine. Altough it works for me, it seems to be not that easy to compile a universal format "one for all"

gereic, 2020-09-27:
Does the battery-Level work with new Version??

bsiege, 2020-09-27:
Yes it works like T-Beam v07. Since both models have the same 100k/100k divider, both have the same offset to real voltage:
```
+------+------+--------+
| Veff | V_gx | offset |
+------+------+--------+
|  3.3 |  3.6 | +0.3   |
|  3.4 |  3.7 | +0.3   |
|  3.5 |  3.8 | +0.3   |
|  3.6 |  4.0 | +0.4   |
|  3.8 |  4.2 | +0.4   |
|  4.2 |  4.7 | +0.5   |
+------+------+--------+
```
I have some Multimeters and a quite good Power Source to get comparable results.
Later i will do same readings with my formula..

gereic, 2020-09-27:
Created new firmware.
Now there is always a firmware with/without psram.
Can you try it ?

bsiege, 2020-09-27:
Yes. that works! Good solution with the two binaries. There seems no other solution.
I found another TTGO Lora32 2.0 which was my first Airwhere setup with external gps. Running out of the box but GPS TX/RX is different again. Will try a setup and give you the data if successful.

P.S. The splitting in python works only for WIN, made a change that works on macOS/Linux too..
Google said "import os" and use "os.path.sep" :-)

```
import os
Import("env", "projenv")
from shutil import copyfile

def copy_file(*args, **kwargs):
    print("Copying file output to project directory...")
    target = str(kwargs['target'][0])
    savename = target.split(os.path.sep)[-1]   # name of environment
    platform = target.split(os.path.sep)[-2]
    print(target.split(os.path.sep)[-1])    
    print(target.split(os.path.sep)[-2])    
    print(target.split(os.path.sep)[-3])    
    savefile = 'bin/firmware_{}.bin'.format(platform)
    #print(savename)
    print(savefile)
    copyfile(target, savefile)
    print("Done.")

env.AddPostAction("$BUILD_DIR/partitions.bin", copy_file)
env.AddPostAction("$BUILD_DIR/firmware.bin", copy_file)
env.AddPostAction("$BUILD_DIR/spiffs.bin", copy_file)
```

Now as the PSRAM issue is addressed and solved, i think i close this issue. Thanks!

closed 2020-09-27 by bsiege