gereic/GXAirCom #4: send groundstation location in OGN groundstation mode

nepat, 2020-09-25:
Have not been able to spot my GXAirCom groundstation on the OGN maps (gliderradar). I believe this is because GXAirCom probably doesn't send a groundstation beacon to OGN. The default of the Raspberry implementations seems to be to send a groundstation beacon every five minutes. The protocol is described here: http://wiki.glidernet.org/wiki:ogn-flavoured-aprs
A groundstation beacon would be great, as this would enable OGN range analysis tools to plot the receiver range of GXAirCom.

gereic, 2020-09-26:
When you want to use the module as ground-station, you have to select ground-station in settings and insert a valid lat and lon.
lat and lon has to be filled manually. It is then not taken from GPS.
Also use the right name according to the specification of OGN.

nepat, 2020-09-26:
The groundstation location and name parameters are actually all set in the GXAircom config. The base station name is set to Huenen2. However over on the OGN aprs servers this arrives as FNB0728B8. GXAircom can be observed reporting to one of the five OGN aprs servers. The process requires trying (opening the web pages of) all five servers linked at http://wiki.glidernet.org/about and then searching for FNB0728B8

gereic, 2020-09-26:
I will check that.
As far as I can see is, that the name is always "FNBXXXXXX" where XXXXXX is the FANET-ID.
I can change that and will write you, when it is ok.

nepat, 2020-09-26:
That would be great. If you look at OGN groundstations over at gliderradar.com, click on the station name and then select 'Raw data', then the datastream will show that the station reports (at least) its name and position to OGN in intervals of five minutes. By doing so, the station then appears on the OGN ground station maps. Having visibility of the ground station in that way is very useful, as it allows to perform remote diagnostics of the performance, range plots, etc.

gereic, 2020-09-26:
ok. New Version is online.
you can check it also on https://glidertracker.de/
I think on https://www.gliderradar.com/ it takes 15-30min. until it shows the station.

gereic, 2020-09-26:
please give me feedback, if it is working as expected.

nepat, 2020-09-26:
With the new version the name of the station is now showing with the chosen name. The position of the stations on glidertracker.org are also shown in the places of the coordinates. However on gliderradar.com the two GXAirCom stations that I am testing (Huenen2 and Huenen 8) are not showing up.

nepat, 2020-09-27:
.......actually now the groundstations are also showing up on gliderradar.com

gereic, 2020-09-27:
Maybe the ground-station is only showing, if it receives the first time an aircraft.

nepat, 2020-09-28:
No, I don't believe that is the case. A 'classic' OGN groundstation built on a raspberry pi will show up in a new location on gliderradar almost instantaneously. This is the case even if the station is receiving no traffic. I was watching what little FANET traffic there was in the air yesterday and some traffic actually got uploaded to OGN by the GXAircom groundstation as I was watching the OGN live map. Strangely however with the latest update of the GXAircom groundstation the groundstation now even shows up on ognrange.glidernet.org. However the traffic data that GXAircom reported to OGN is not shown on ognrange.glidernet.org. These are only 'cosmetics', as the main function (uploading traffic to OGN) is clearly working fine.

nepat, 2020-10-04:
One difference that it did notice today in GXAirCom position reports on OGN is that the position doesn't include the signal strength (in db) with which the position was received.
Also: if a target was showing on gliderradar.com and was being received on OGN by my GXAirCom groundstation and I clicked on the target, then very briefly my GXAirCom groundstation would actually show up on the gliderradar.com map.

nepat, 2020-10-06:
Gerald, great to see you had success with 1.5.3 !

closed 2020-10-06 by nepat

gereic, 2020-10-06:
nepat
Can you check if new Version is working and give me some feedback ?