gereic/GXAirCom #5: connect to WiFi: always try to reconnect

nepat, 2020-09-28:
The menu choice connect to wifi:connect (always try to reconnect) can't be saved persistently. It will always returns to 'try only once'.

gereic, 2020-09-28:
Bug fixed with new Version

nepat, 2020-09-28:
Thanks ! Is the filesize for firmware_esp32noPSRam.bin correct ? At 192 kb the file seems very small compared to previous versions (or the PSRam version)

gereic, 2020-09-29:
just created a new bin-file. Filesize is 1.6MB

nepat, 2020-10-03:
Thanks for uploading the file. Recently I had the groundstation connected to a temporary location (a hotel). When I returned home and tried to access the groundstation via WiFi (in order to change the Wifi settings back to my home location again), I failed to connect to the groundstation over WiFi. Eventually I had to erase and reflash the firmware. Could it perhaps be that the groundstation was so busy trying to reconnect to the (no longer present) hotel network, that there was no capacity to allow WiFi connections to the groundstation ?

And a follow-up question: is it possible to reset the settings to default without having to re-flash the board ?

danielvelezbravo, 2020-10-03:
nepat, i think you are into something... On previous versions If i set a wifi hotspot to use on airmode, and the wifi wasn´t available, then it was impossible to connect to the access point for settings and so.  That´s why gereic made the "try to reconnect once or try to reconnect always" available.   So if no success trying to reconnect, then the wifi config was available once more.  

Gerald, maybe we (you) need to change the "always reconnect" for a "5 minutes" reconnect timeout so we can regain control of the device... or maybe a Two button long push (like power and settings button) so that it disables the "access point credentials" or makes it "try to reconnect once" only.    

Like: If on the "try to reconnect loop" then "settings button long press behaviour = Set reconnect to "none".

nepat, 2020-10-03:
In groundstation mode the station should recover automatically from possible failure scenarios. From a user perspective it would be sufficient if the WiFi connect would happen 'eventually'. By that I mean that for a groundstation it would be perfectly acceptable if WiFi reconnect attempts would happen only every couple of minutes. That would be sufficient for the purpose of eventually reconnecting after the WiFi connection has gone down.

A side question: am I right in guessing that GXAirCom will reboot once a day ? .... if it doesn't then a regular reboot would also be an additional useful longterm stability feature.

gereic, 2020-10-06:
Just uploaded new Version V1.5.3, where wifi-reconnect-timeout is now 10min. This should be enough time to connect to GxAircom and adjust settings.

nepat, 2020-10-07:
Hi Gerald - thanks for the recent updates. I tested as follows: connected GXAirCom to my mobile phone WiFi hotspot. Checked that GXAirCom as working fine with the hotspot. Turned off the hotspot. Unplugged and restarted GXAirCom. Tried to connect to GXAirCom WiFi with a laptop computer. During the first about two minutes the connection was refused. Then however I managed to connect and change the GXAirCom WiFi settings to another WiFi network. I would consider this a successful test of the 'change location' scenario. Another minor WiFi detail: I can connect with a Windows Notebook and an Android tablet. With another Android phone (Xiaomi Redmi Note 7) WiFi will connect, but then immediately disconnect again. This has been the case with all releases of GXAirCom. I wonder whether other users are also seeing this behaviour. BR Patrick

closed 2020-10-07 by nepat