gereic/GXAirCom #6: dB signal strength in OGN data packets

nepat, 2020-10-08:
Today I was watching out for dB signal strength of targets on glidertracker.de. They showed up with zero dB (see screenshot). Something must be wrong. Does OGN perhaps not like the triple digit figures that GXAircom is reporting ? On OGN receivers based on an SDR a great signal would probably be somewhere around 50dB, but never in the triple digits.

[image]

gereic, 2020-10-09:
the db-signal on OGN is not the same as on GXAircom.
on it is a SignalNoiseRatio in DB.
on GxAircom it is the signal-strength.

But anyway. In my configuration it was showing the db.
Can you try it, when a Ground-station and a Air-module are very close, to see if it never gives a value or if the value is incorrect.

nepat, 2020-10-09:
You are right. Today when I was observing traffic about 6Km away (line of sight) that traffic all registered with 0dB.

When I set up a TT-BeamV1.1 for test purposes 3m from the antenna of the receiver the reading was 9.7dB on glidertracker.de and -55dB on the screen of the Heltec WiFi Lora board receiver. When I repeated the test with a Skytraxx vario from about 15m distance the reading on glidertracker.de also did not go above 9.8dB. I can't say what the Heltec board showed at that moment, as the Skytraxx required me to keep a certain speed of motion to keep transmitting data.

nepat, 2020-10-09:
In the same position a 'classic' OGN receiver with RLT-SDR stick and exactly the same antenna reports the TT-Beam V.1.1 with 29dB on glidertracker.de.

gereic, 2020-10-11:
Created a new Version, where SNR-Value is now calculated from RSSI.
It is not the same, but as in the documentation from Lora the SNR-Value goes from -10dB to 20dB.
So I decided to use the RSSI-Value which goes from -120 to 0.
calculation is now 120 + rssi-value --> so we got 0 for very poor signal and 120 for best.
I tested in my home with a distance of 10m and got -79dB on Oled and 41 on glidertracker.de.
So I think this is a more usefull value then using snr for the sx1276

nepat, 2020-10-11:
[image]

Today the new version showed more realistic values. I used a tiny antenna from an indoors position. 8db is therefore quite good.

One last remaining OGN issue: I can't see range plots on https://ognrange.glidernet.org/ for the targets received by GXAirCom

closed 2020-10-26 by gereic