gereic/GXAirCom #7: 1.6.0

nepat, 2020-10-25:
two issues on 1.6.0 and one issue from the previous version:

db reporting:
- db numbers in OGN now just show as zero

Settings;
Updated a T-Beam T22 V1.1 from the previous version. The unit had been configured as an air- unit. After the update the settings were no longer present. The pilot field for example was empty, but the header of the webpage still showed the pilot name that had been present in the settings of the previous version. Re-entering the pilot name and operating mode (air-mode, OGN tracking) doesn't work. After saving and rebooting the settings are not retained. Also the entire 'info' screen lacks any information. Also tried physically to physically restart (unplug power) the unit - with the same effect.

Previous version - occasionally stops reporting traffic
Occasionally the unit would loose connection to the OGN. The station would disappear from the station map and cease reporting traffic. When I returned to the unit everything on the outside looked normal. The display was on and I could also access the unit via the webpage. A reboot then brought everything back to normal again. Strangely, during on of these dropouts the base station was still updating heartbeat packets to the airwhere map website and was transmitting out FANET status packets (these were picked up by neighbouring OGN receiver).

nepat, 2020-10-25:
The settings issue that is reported above doesn't seem to affect the operation of the device itself. I can still see the tracker in the OGN and it is seemingly still working with the 'previous' settings. The issue is just, that the settings can no longer be accessed or changed over the web page. It seems like the actual settings stored on the device and the webpage have become 'disconnected'.

gereic, 2020-10-26:
Uploaded new Version 1.6.1 please try it.
According, the settings-issue, please do a complete flash.

nepat, 2020-10-26:
Thanks ! 1.6.1 fixed the db numbers in OGN. The update over the web interface went smootly and after the update all the previous settings were back to 'normal'. A complete flash was not necessary.

closed 2020-10-26 by nepat