gereic/GXAirCom #8: 1.7.0

nepat, 2020-10-30:
Did a bit of testing:
In OGN groundstation mode receive strength (db) is not transmitted to OGN
A 0.2.6 version SDR OGN groundstation (this is a Flarm only groundstation) will not receive a GXAirCom tracker sending Legacy signals. Whether Legacy really works should be verified against a true Flarm receiver - for example an XCTracer vario or glider
In groundstation mode targets were not showing up on the Airwhere map.

gereic, 2020-10-31:
V1.7.1
fixed (again) bug with receive-strength (db).
You have to switch on Live-tracking service also in Ground-station-mode
So If you want that ground-station is broadcasting to Airwhere you have to switch on Live-tracking in settings.
This is essential for all live-tracking-services.
So you can decide, if your ground-station should broadcast to which service.

nepat, 2020-10-31:
Installed, tested, works great. Thanks !

closed 2020-10-31 by nepat