gereic/GXAirCom #9: GPS fix on LILYGO TTGO T-Beam V1.1

nepat, 2020-11-19:
The GPS lock performance on the LILYGO TTGO T-Beam V1.1 (SX1272) can be unpredictable. Often the GPS light on the PCB will blink, but the web interface will not show a GPS lock. Strangely a reboot on the web interface will then predictably result in a GPS position lock being established. This makes it a bit difficult to use as a tracking device, as one can never be certain whether a GPS position has been successfully established. It can't be the antenna hardware, as connecting an active antenna doesn't change the GPS behaviour. Only a reset of the device after a GPS lock has been established will always work.

gereic, 2020-11-20:
Can you check the output on the serial-port if there are NMEA-sentences ?

nepat, 2020-11-20:
I didn't have the serial output activated at all. Perhaps that was the issue. I will watch and report back if the issue appears again. By the way: Connecting a tiny active ceramic antenna made a big difference in the GPS signal reception. With the included antenna it would sometimes take forever to find GPS satellites. Now with the active antenna a fix can even be obtained indoors.

closed 2020-11-20 by nepat