gereic/GXAirCom #10: Course and Speed being reported for OGN groundstation

nepat, 2020-11-20:
When the position is reported by GXAirCom in the 5 minute position beacon signal of a groundstation to OGN, this also includes course and speed. I don't see the other groundstations doing this. Their message format ends with A=XXXXXX
Is this necessary ? Could this perhaps prevent the OGN Range Tool (https://ognrange.glidernet.org/) from recognising GXAirCom as a valid ground station ?

nepat, 2020-11-20:
I mean that GXAirCom reports the following example:
20/11/2020 19:47:07+01:00: GETRONIX>APRS,TCPIP*,qAC,GLIDERN3:/184707h4804.37NI01444.09E&000/000/A=001059 !W44!

whereas the format from other OGN stations would look more like this
20/11/2020 19:47:07+01:00: GETRONIX>APRS,TCPIP*,qAC,GLIDERN3:/184707h4804.37NI01444.09E&/A=001059

what by the way is the !W44! doing at the end ?

gereic, 2020-11-23:
The !W44! is the extension for lat and lon to get more precise data.
see [https://github.com/svoop/ogn_client-ruby/wiki/SenderBeacon](url)

nepat, 2020-11-23:
OK, I see. By the way, when watching targets reporting back to OGN I noticed that the altitude would 'jump' as the target gets handed off between OGN stations running GXAirCom and stations running the SDR software on a Raspberry Pi. This makes the altitude plot look like a sawtooth pattern. I will watch this, but I believe to have a read a thread over on the OGN forum that this also was once an issue with SDR receivers that used different Geoid models.

gereic, 2020-11-24:
I Uploaded a new Version without speed and heading in ground-station-mode.
But I think it doesn't have an affect.
I created a new Topic in the OGN google-group, hope to get help.

nepat, 2020-12-03:
Perhaps it is just that the OGN Range Tool doesn't accept values from FANET stations. This could make sense, as FANET is a repeating protocol and range statements therefore are not necessarily directly received positions. re the Geoid value the configuration on the SDR OGN receiver was actually incorrect - not on the GXAirCom receiver.

closed 2020-12-03 by nepat