gereic/GXAirCom #13: GPS fix with the T-Beam V1.1

nepat, 2020-12-06:
on first start the T-Beam V1.1 will show a blinking GPS light but:
- no GPS position on the webpage
- no GPS position is transmitted over FANET
- no GPS position can be obtained over Bluetooth with LK8000

fix
- enter the webpage
- hit settings save and reboot (with unchanged settings)
- now GPS works as expected

this behaviour can be reproduced in a persistent manner

This seems to be a known issue with the T-Beam and can be fixed with a GPS reset.
https://github.com/Xinyuan-LilyGO/LilyGO-T-Beam/issues/28

gereic, 2020-12-07:
How can i reproduce the error ?
Do I need a new board or do I need to flash it or ....

nepat, 2020-12-07:
It's a perplexing issue. In my tests the board had always been shut off for a couple of days. After sitting just a couple of minutes in the off state the problem doesn't occur. Will wait a couple of days and repeat.

gereic, 2020-12-07:
The thing is, that I know already the problem, that on the V1.1 there is no NMEA-Output and there is already a fix in the program.
If there is still a problem with that I have to check, where the problem is.

danielvelezbravo, 2020-12-07:
I haven´t experimented this issue.  

Usually whe an TTGO T-Beam unit arrives it has Mestastic or SoftRF preinstalled.  I first connect them for a while to let them settle and see that they are working with the pre-configured software (gps reception)...
Then I flash GXAirCom... maybe then, as I want to test the software with LK8000 and android ble connection, to preconfigure it "out of the box" for final users, I enter the settings and set some parameters (BLE, maybe aircraft type). So I´m doing the "walkthrough" every time.  

The thing is that they don´t show the strange behaviour after 2-5 or 10 days out of commission. I just start them up again (like after a week of no flying), and they work flawless.   

Only have trouble with one unit that was laying out for a long time without a battery installed, and it took some time to get a gps fix, but no "gps blinking with no gps position logging" at the momment.

I´m waiting for 5 new boards (TTGO T-Beam V1.1. with oled).  When they come, I will try to reproduce the behaviour.  I´m guessing after first 1.8 flash, just see if i get some gps info on the info webpage, before getting into settings.

nepat, 2020-12-07:
this evening:
- turned on the TTGO T-Beam with the switch (using the internal battery)
- after about 2 minutes the red light started to blink
- even after waiting for about 10 minutes the TTGO T-Beam would not send a position via OGN
- turned the TTGO T-Beam off with the switch
- waited for about 1 minute
- turned on the TTGO T-Beam with the switch (using the internal battery)
- almost immediately the red light started to blink
- immediately started to get position reports via OGN
....and repeated the same process once more successfully
hypothesis:
- using the switch to restart seems to have the same effect on GPS reception as rebooting via the web interface
- a cold reboot (I had waited about 10 hours) causes the GPS on the TTGO T-Beam to fail to achieve a GPS lock, whereas a warm reboot resolves the issue

Will try tomorrow whether there is any difference between (internal) battery and external USB power.

gereic, 2020-12-07:
A serial-output would be nice, but I don't know how you can do that.
in Serial-Monitor at startup, there you see if the software has found a ublox GPS. 
If so, it sets the NMEA-OUTPUT

danielvelezbravo, 2020-12-07:
> this evening:
>
> * turned on the TTGO T-Beam with the switch (using the internal battery)
> * after about 2 minutes the red light started to blink
> * even after waiting for about 10 minutes the TTGO [...]

So, to short out the issue.....  How is your TTGO reporting to OGN?  By being received by a base station that forward it to ogn, or directly by connecting to the wifi at your place?   
And does the navigation software on your phone (lk8000, xctrack, xcsoar) detects the TTGO Flarm device or do you need to "reboot" in order to being recognized (broadcast its own position) to the android software?

nepat, 2020-12-08:
re: How is your TTGO reporting to OGN?
--> I receive via 3 own base stations (2 GXAircom, 1 SDR receiver) - but that is not the point as a missing position report on OGN also coincides with not GPS lock being shown on the webpage of the TTGO
re: And does the navigation software...
--> Yes, I have to reboot the TTGO before a position is reported to the android software over bluetooth

Just tested with external USB power. This makes no difference.

gereic, 2021-01-04:
Hi nepat,
Can you try version V2.1.0

1. goto settings
2. press "config-GPS"
3. wait 10econds.

Then try if error still occurs.
The config-GPS-Button does the following:
reset GPS to factory-defaults
enable NMEA-output to uart1
disable not needed NMEA-sentences and enable GPGGA and GPRMC-sentences.

nepat, 2021-01-07:
I tried several times with 2.2.0.  Each time the GPS acquired a fix very rapidly. Also from a cold start (waiting one day between tests).  I never had to use the config-GPS button. Thank you for the fix !

closed 2021-01-07 by nepat