gereic/GXAirCom #15: T-SIM7000 as am - connection redundancy

BPieter, 2020-12-16:
Hi,

Would it also be possible to use the SIM7000G as a tracker? Or is this already tested?
Using the functionality it should be possible to both have FANET signals sent over LoRa while at the same moment sending data to a traccar server. Offering connectivity and tracking in more situations or conditions.

Sincerely

gereic, 2020-12-16:
Hi,
It can act as OGN-Tracking-server already.
If I should extend it to traccar-server, I need the traccar-server-address, to implement and test.
Please register the device "FANET_075890"

BPieter, 2020-12-16:
Maybe I understand you wrong, what do you mean with a OGN tracking server?
Normally I flash gs_sim7000_psRam on T-SIM7000G and am_oled_psRam on T-BEAM.
So can I set operating-mode already to AIR-Module on T-SIM7000G.

Also I don't know traccar yet, why register a specif FANET_075890 id?
Would it not work just by setting traccar server? I will go through documentation already.

BPieter, 2020-12-16:
Hi,

traccar server: lx.locrepo.be 
device registered

If I add the T-BEAM device ID as you describe, it is indeed passed to the traccar server by the T-SIM7000G ground station.
However, my goal would be to have it send directly by the tracker either by using WiFi (mobile as hotspot when flying) or using mobile connection (if using T-SIM7000G as tracker)

What is your opinion?

closed 2021-12-26 by gereic