gereic/GXAirCom #16: failure to wake up from sunset to sunrise sleep

nepat, 2020-12-18:
Last night I set up three boards (TTGO 1.0, TTGO 1.3 and T-Sim7000G) on 1.9.1 to go to sleep from sunset to sunrise by using the respective option. They all failed to wake up the next morning. The are no longer registering on OGN. They don't show up on Wifi.
All are powered over USB. In the the T-Sm7000G the battery was in and the board was connected to USB power.
Other observations.
On the TTGO 1.0 the red led on the mainboard is blinking as usual. However the chipset feels cold. When operating and online, the chipset would be warm.
On the TTGO 1.3 the green led on the board is on as usual. However the chipset also feels cold.
The same applies to the T-Sim 7000G.
Other observations: with 1.9.0 and 1.9.1 the boards would not automatically reboot after an update of the firmware over the web frontend. The boards appeared to 'hang' and required a hard reboot (by pulling USB power). The same also happenend to me wiht the TTGO 1.0 after I changed the sleep setting on the website.
In another test on 1.9.0 the T-Sim 7000G would not start without a battery in the battery compartment. Night sleep mode was not enabled for this test. Power draw over USB was 7.5mA, which would be typical for a deep sleep mode.
To complete the diagnostic picture I will now also run the T-Sim 7000G for a night just on battery power.

nepat, 2020-12-19:
Just on battery power the T-Sim 7000G will also not wake up after night sleep.

gereic, 2020-12-19:
Found a bug in calculation.
Fixed it, but not full tested.
can you test it ether with you devices ?

gereic, 2020-12-19:
one thing, when you flash the controller it is min. 3min awake.
This is for changing the setting when no daytime (otherwise you have no chance to switch settings when you reset the device).

nepat, 2020-12-19:
Thanks for the update. 3 devices loaded with the new version. Waiting for tomorrow morning.

nepat, 2020-12-20:
The wakeup this morning was successful. My observation is indirect via the 'raw data' section on gliderradar.com. It shows an interesting pattern. First the station send one single position packet, then waits for some time before resuming the regular 5 minute pattern of sending position packets. For the three stations the pattern was: Station 1 07:33, 08:09; 07:26, 08:09; 08:01, 08:20
Waiting for them go go to sleep this evening.

nepat, 2020-12-20:
All devices went to rest at around about the same time and at the time when calculated sunset would occur at their location. Now waiting again for tomorrow morning.

nepat, 2020-12-21:
Station 1 07:46, 08:10; Station 2 08:05, 08:10, 08:14; Station 3 07:09, 08:10
Working Great. Thank you !

closed 2020-12-21 by nepat