gereic/GXAirCom #18: T-SIM7000G - reconnect after battery depleted

nepat, 2020-12-29:
The test scenario:
T-SIM7000G in sleep sunset to sunrise mode
let the battery deplete until the device no longer reports to OGN
plug in USB to recharge the battery

Symptoms:
even after several days with USB connected and the battery charge light showing a full charge, the device would not reconnect to OGN
after disconnecting USB and letting the device sit for 19 hours still no connection to OGN

Observations:
the 'Status' LED next to the Sim Antenna lights up solid red
the 'Netlight' LED next to the 'Status' LED blinks slowly (about every two seconds)

After removing and re-inserting the battery everything is back to normal. The device connects to OGN and the 'Netlight' LED blinks at a fast pace (I didn't try the power switch).

nepat, 2021-01-10:
The issue persists also in testing with 3.0.0. The battery will charge back up both over USB (tested yesterday) as also over Solar (tested today). However the T-SIM700G will not come back online despite the battery being full (power LED shows green). All LED next to the SimCom Chip stay black. The only way to revive the device is a) to remove the battery or b) to cycle the physical on/off switch. Perhaps: the onboard power logic has a design bug ? the software is not able to sense a charged battery ? the data modem crashes on low power and needs a reset ?

gereic, 2021-01-10:
When you plug in USB and start serial-monitor, do you have any output ?

nepat, 2021-01-10:
OK, will report the findings with the next recharge cycle.

nepat, 2021-01-13:
No, there is not output on serial-monitor. The board will only come back to life after cycling the on/off switch or removing the battery. Looks to me like a possible hardware design flaw. Will in the future test whether it is possible to bypass the internal power circuit electronics by connecting a battery to the battery pins of the board.

gereic, 2021-01-13:
very strange,
I tried it also many times and it works every-time.
When battery was completely empty I replaced it and all was starting normal.

danielvelezbravo, 2021-01-14:
There is also a similar behaviour on the TTGO T-Beam 1.1.  When you "flash"
a new firmware and spiffs, via pc, the unit gets frozen. Then you must
recycle the power button twice. If you hold the power button 10 secs, the
unit (screen) will turn off but there would be a blue led still on. Then
you start once more the unit by short press on the power button (1 - 2
secs). Then you wait 10 secs, and then power down again (hold 10 secs at
least) and the unit will fully power off (screen off and blue led off.

nepat, 2021-01-14:
I have started a test whereby the battery is connected to the VBAT terminal of the board. According to the documentation of the board on Github, that should bypass the power switch. https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G

nepat, 2021-01-15:
Test Update: With the battery connected to the VBAT terminal the following happened. The battery depleted and the T-SIM7000G shut down. Then I charged the battery with an external charge circuit that is connected in parallel with the battery. Charge started with about 75mA and remained at that point for about 1.5 hours. The T-SIM7000G at that time remained 'dark' with only a blue LED on and the ESP32 chip being cold to the touch. Then, after about 1,5 hours the board returned back to service. Right now in the web interface the battery voltage is showing 3.45 V. If at this moment I disconnect the USB charger, the board doesn't seem to operate in a stable manner yet, as it starts to do re-starts (the red system and network leds go dark and then start up again, the blue led turns on) as soon as I disconnect the USB charger.

nepat, 2021-01-17:
Test Update: With the external battery connected, the board can be charged over both the USB connection as well as the solar charge port. Conclusion: users who want unattended operation on solar are probably better advised to connect an external battery and thereby bypass the internal switch circuit.

closed 2021-01-17 by nepat