gereic/GXAirCom #19: FANET temperature shows too high (double) value

nepat, 2020-12-30:
The test setup: LILYGO® TTGO ESP32 SX1276 LoRa V1.0 with BME280 temperature/pressure/humidity sensor. No other sensors are connected. https://www.aliexpress.com/item/32845370112.html?spm=a2g0s.9042311.0.0.27424c4dzkWaJT

The issue: on the webpage of the TTGO the temperature value and all other values are shown correctly. However the temperature value that is received via FANET (viewed on the gliderradar.com website) is exactly double the correct temperature. The other values (humidity, pressure) that are received over FANET are correct.

Another small issue: as soon as I set a temperature  (K) offset, the board website starts to show extreme (many digit) figures.

gereic, 2020-12-30:
Where can you see the Temp on gliderradar.com ?
Can you check the temp on LK8000 connected with air-module GXAircom ?
There you should see a weatherstation and in the info-page you find all values (temp, pressure, wind, ...)

gereic, 2020-12-30:
OK, I see the weather-station on OGN.
Will double-check that with LK8000

gereic, 2020-12-30:
I checked it with my LK8000 and all values there were displayed correctly.
I wrote a email to the OGN-Team. Think it is a problem on OGN-Receiver.

nepat, 2020-12-30:
Your own stations report temperature correctly on OGN. Do your other stations also report the temperature from the BME280 ?

gereic, 2020-12-30:
What do you mean, with "your own stations" ?
GXAircom doesn't send weather-data to ogn at the moment.

gereic, 2020-12-30:
But when sending weather-data to FANET, the OGN-Stations will receive that data and push it to OGN I think.

nepat, 2020-12-30:
That's right. The FANET data is received by OGN stations and then forwarded to OGN. By 'your own stations' I mean the GXAirCom stations around Amstetten. Their weather data is also picked up by OGN stations and then forwarded to OGN. I just added my own station to Windy. That should provide another verification on how data is reported to the outside. Apparently it takes a long time until a station first shows up on Windy.

nepat, 2021-01-01:
Update: Windy reports the correct values and works very nicely. It takes about a day until Windy starts showing values.
A minor observation: I did have two instances on Windy where briefly extremely high humidity (121318%, etc.) values were reported. I don't know the cause. Perhaps low power on the sensor or another hardware malfunction. Perhaps it could be considered to have the software reject values that can't make sense because they are outside of the range of conditions existing on earth.  That way a malfunctioning sensor will not cause data pollution.

nepat, 2021-01-01:
The doubling of the temperature values also occurs if weather data is re-broadcast over FANET from Weather Underground.

nepat, 2021-01-02:
On the humidity issue with the BME280: the sensor doesn't seem to like prolonged exposure to high (condensing) humidity and then gets 'stuck'. Ignoring the humidity by just reporting 100% until the sensor becomes 'unstuck' again should work for practical purposes.

nepat, 2021-01-02:
In the Interlaken area in Switzerland there are a couple of windline.ch stations that also report weather to FANET. Their data shows the same behaviour. Temperature on gliderradar.com is shown with a multiple of 2 of the temperature that is shown on the windline website for the respective device. The multiplication must therefore happen somewhere along the path at OGN or gliderradar.com

closed 2021-01-02 by nepat