gereic/GXAirCom #20: unable to install on a TTGO T3 V1.6

nepat, 2021-01-02:
On the TTGO T3 V1.6 the versions prior to the AM/GS version split will install. However newer versions will not start (both if flashed over USB or if updated over the web interface).

gereic, 2021-01-03:
Which version have you flashed ?
Can you post a console-log ?

nepat, 2021-01-03:
I flashed 1.8.0_gs_oled_nopsRam and got the following:

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
[W][esp32-hal-psram.c:30] psramInit(): PSRAM init failed!
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40134cd4: 40010441 01414554 54400100
Core 0 register dump:
PC      : 0x40134cd8  PS      : 0x00060030  A0      : 0x80100c46  A1      : 0x3ffbbc80  
A2      : 0x8001f880  A3      : 0x8001f880  A4      : 0x40088d04  A5      : 0x00000001  
A6      : 0x00060c20  A7      : 0x00000000  A8      : 0x800ff91c  A9      : 0x3ffbbc60  
A10     : 0x00540000  A11     : 0x00000001  A12     : 0x3ffca625  A13     : 0x00000001  
A14     : 0xffffffff  A15     : 0x00000000  SAR     : 0x0000000f  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe  

Backtrace: 0x40134cd8:0x3ffbbc80 0x40100c43:0x3ffbbca0 0x40104223:0x3ffbbcd0 0x40091f39:0x3ffbbcf0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
[W][esp32-hal-psram.c:30] psramInit(): PSRAM init failed!
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40134cd4: 40010441 01414554 54400100
Core 0 register dump:
PC      : 0x40134cd8  PS      : 0x00060030  A0      : 0x80100c46  A1      : 0x3ffbbc80  
A2      : 0x8001f880  A3      : 0x8001f880  A4      : 0x40088d04  A5      : 0x00000001  
A6      : 0x00060c20  A7      : 0x00000000  A8      : 0x800ff91c  A9      : 0x3ffbbc60  
A10     : 0x00540000  A11     : 0x00000001  A12     : 0x3ffca625  A13     : 0x00000001  
A14     : 0xffffffff  A15     : 0x00000000  SAR     : 0x0000000f  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe  

Backtrace: 0x40134cd8:0x3ffbbc80 0x40100c43:0x3ffbbca0 0x40104223:0x3ffbbcd0 0x40091f39:0x3ffbbcf0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
[W][esp32-hal-psram.c:30] psramInit(): PSRAM init failed!
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40134cd4: 40010441 01414554 54400100
Core 0 register dump:
PC      : 0x40134cd8  PS      : 0x00060030  A0      : 0x80100c46  A1      : 0x3ffbbc80  
A2      : 0x8001f880  A3      : 0x8001f880  A4      : 0x40088d04  A5      : 0x00000001  
A6      : 0x00060c20  A7      : 0x00000000  A8      : 0x800ff91c  A9      : 0x3ffbbc60  
A10     : 0x00540000  A11     : 0x00000001  A12     : 0x3ffca625  A13     : 0x00000001  
A14     : 0xffffffff  A15     : 0x00000000  SAR     : 0x0000000f  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe  

Backtrace: 0x40134cd8:0x3ffbbc80 0x40100c43:0x3ffbbca0 0x40104223:0x3ffbbcd0 0x40091f39:0x3ffbbcf0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
[W][esp32-hal-psram.c:30] psramInit(): PSRAM init failed!
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40134cd4: 40010441 01414554 54400100
Core 0 register dump:
PC      : 0x40134cd8  PS      : 0x00060030  A0      : 0x80100c46  A1      : 0x3ffbbc80  
A2      : 0x8001f880  A3      : 0x8001f880  A4      : 0x40088d04  A5      : 0x00000001  
A6      : 0x00060c20  A7      : 0x00000000  A8      : 0x800ff91c  A9      : 0x3ffbbc60  
A10     : 0x00540000  A11     : 0x00000001  A12     : 0x3ffca625  A13     : 0x00000001  
A14     : 0xffffffff  A15     : 0x00000000  SAR     : 0x0000000f  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe  

Backtrace: 0x40134cd8:0x3ffbbc80 0x40100c43:0x3ffbbca0 0x40104223:0x3ffbbcd0 0x40091f39:0x3ffbbcf0

Rebooting...
ets Jun  8 2016 00:22:57

nepat, 2021-01-03:
Would a (not yet existing) PSRAM image be required to make this board work as a groundstation ? This (a PSRAM GS image) would be useful anyway also for the T-Beam boards.

gereic, 2021-01-04:
Can you try newest Version "firmware_v2.0.1_gs_oled.bin" ?

nepat, 2021-01-04:
I just did the following tests:

1.6.1_esp32noPSRam.bin worked
1.6.1_esp32psRam.bin  didn't work

2.0.1_gs_oled.bin didn't work
2.0.1_am_oled_psRam.bin didn't work

Updating over the webpage from 1.6.1_esp32noPSRam.bin to 2.0.1_gs_oled.bin resulted:
- the screen continued to show "FW-Update wait..."
- the device disappeared from Wifi
- after unplugging from USB the device didn't restart

Below is the console outpout with 2.0.1_gs_oled.bin flashed over the ESP32 Download Tool

Backtrace: 0x40177f6c:0x3ffbbc80 0x40136e1a:0x3ffbbca0 0x401061df:0x3ffbbcd0 0x40091f39:0x3ffbbcf0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
[W][esp32-hal-psram.c:30] psramInit(): PSRAM init failed!
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40177f68: 40415540 00000100 51044105
Core 0 register dump:
PC      : 0x40177f6c  PS      : 0x00060030  A0      : 0x40136e1d  A1      : 0x3ffbbc80  
A2      : 0x8001f880  A3      : 0x8001f880  A4      : 0x40088d04  A5      : 0x00000001  
A6      : 0x00060c20  A7      : 0x00000000  A8      : 0x801018c0  A9      : 0x3ffbbc60  
A10     : 0x44150000  A11     : 0x00000001  A12     : 0x3ffca671  A13     : 0x00000001  
A14     : 0xffffffff  A15     : 0x00000000  SAR     : 0x0000000f  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffd  

Backtrace: 0x40177f6c:0x3ffbbc80 0x40136e1a:0x3ffbbca0 0x401061df:0x3ffbbcd0 0x40091f39:0x3ffbbcf0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac

gereic, 2021-01-04:
The problem is, that I don't have a Board for testing.
How many boards of this type do you have ?
Can you send me one board ? (Austria)

nepat, 2021-01-04:
Sure, I have a spare one. Letter will be posted tomorrow.

gereic, 2021-01-08:
Bug fixed in Version V3.1.0
Can you double-check ?

gereic, 2021-01-08:
version is online

nepat, 2021-01-09:
Thanks Gerald ! With the update the TTGO T3 now works fine. With the built-in SMA socket the T3 right now is my favourite board for use in a ground station.

closed 2021-01-09 by nepat