gereic/GXAirCom #26: Trying to build in VCS via PIO

parahooners, 2021-02-04:
Hey,

Getting and issue with web access after uploading via PIO on Mac M1 

E][WebHelper.cpp:565] onPageNotFound(): [192.168.4.2] HTTP GET request of /favicon.ico

An ideas....

parahooners, 2021-02-06:
Its the Platform.ini file, its set for 

[esp32_base]
framework = arduino
board = esp32dev
platform = espressif32

is there a workaround for the TTGO T Beam board. ?

[env:ttgo-t-beam]
platform = espressif32
board = ttgo-t-beam
framework = arduino

parahooners, 2021-02-06:
Being on Mac M1 I'm unable to run the esp32 Download tool, have tired on a VM but the ARM system in the win 10 insiders is not accepting to Com port at the moment.....

gereic, 2021-02-08:
esptool.py should work on mac.
esptool.py -p %PORT% -b 921000 write_flash 0x1000 bootloader_dio_40m.bin 0x8000 partitions.bin 0xe000 boot_app0.bin 0x10000 %firmware% 0x3d0000 spiffs.bin

replace %PORT% with the port of the controller (don't know how you can find that out)
and replace %firmware% with the right firmware-file for your controller

parahooners, 2021-02-08:
Thanks @gereic that worked a treat...

closed 2021-02-08 by parahooners