gereic/GXAirCom #28: Oled TTGO T Beam

parahooners, 2021-02-09:
Hello,

Okay so the new PIO file is working with out errors...

Just one issue regardless if I flash Via Esptool.py or Platform IO the Oled just won't stay in the setting in the Web page.... every thing else saves and sticks just the oled won't...

Have tried a few thing but it just won't stick..

Iv set the Oled as per the pinout

parahooners, 2021-02-09:
@gereic 

Scanning the IC2 I can see the screen is there and working ?

21:31:58.123 -> I2C device found at address 0x34  !
21:31:58.156 -> I2C device found at address 0x3C

parahooners, 2021-02-09:
Think its an issue with the Webhelper file...

Changed the define in main.h file for old 96 to 1 rather and 0 and swapped no_display to 1 and bingo screen is on...

the webhelp can't be uploading right....

gereic, 2021-02-10:
Found a Bug when switching to OLED.
please try Version v4.3.4

closed 2021-02-10 by gereic

parahooners, 2021-02-10:
@gereic can confirm fix it working