gereic/GXAirCom #31: Gitter Community

Martenz, 2021-02-19:
Hi,

I'm playing a bit with your code and I would like to try to help if I can or at least contribute.
I started actually with the softrf project before but from what I see your project is more close to paragliding.

I'm testing a bit of all hardware, just received the heltec esp32 v2 and I do have a T-Beam V1.1.

I do have some questions any chance we can have a chat or is there a Gitter room to discuss about developments?

Thanks,

Martino.

danielvelezbravo, 2021-02-19:
Chat here....if simple enought maybe i can jump in and help.

Martenz, 2021-02-20:
Thanks a lot,

So first question to understand if I got it right. 

GxAir is able to transmit and receive through the fanet protocol so it should be compatible to be seen and receive from other devices like skytraxx or in general equipped with fanet.

Here where I live there is a quite extense use of flarm and just couple of ogn receivers but no fanet, from the code I see in this repo there is some flarm code..
I would like to understand if it is possible to send also flarm nmea data?

Would it be possible to integrate basically this device with a sort of fanet+ to be seen both from fanet and flarm devices? Or if not possible, what if we can put together two antennas one sending and receiving flarm compatible data that is seen also from ogn (like legacy option in softrf) and fanet on the other ? And collect received data forwarding to serial, bluetooth or WiFi in single data .. because the only other solution I see now is to have two devices  but some app like xctrack accept only one sensor at a time.

Thanks,

Martino

gereic, 2021-02-20:
GxAircom is already able to transmit and receive Fanet, so you will see and you will be seen by other Fanet-devices like Skytraxx.
It is also able to transmit legacy-protocol like softrf. But only to transmit.
So it sends Fanet and legacy and can receive only Fanet.
The OGN-Stations can also receive Fanet (if software-versiond is minimum 2.8)
So if your friends have a skytraxx-device (or other Fanet-device), the can see you and you can see them.
If your friends have a Flarm-device, the can see you but you can't see them.

Martenz, 2021-02-20:
> GxAircom is already able to transmit and receive Fanet, so you will see and you will be seen by other Fanet-devices like Skytraxx.
> It is also able to transmit legacy-protocol like softrf. But only [...]

Thanks a lot for the detailed explanation! So basically GxAircom is a fanet+ right? fanet and legacy (or flarm compatible) like the skytraxx from what I understand. 

Do you think it would be possible to receive also flarm or legacy in the future? I mean I can use an  other small STM32 Dongle with softrf and forward data to lk8000 but for xctrack I can use a sensor at a time.. I can even use a GxAir a Softrf and an esp32 to  grt both serials and forward everything to bluetooth in a single connection.. but seems a bit too bulky..

Tomorrow I'm going to test a heltec ground station to see my friend skytraxx, next week I should receive an other T-Beam and install and test latest GxAircom thanks.. Can I build and flash directly from platformio with the git ? is there an how to or some setting to build and test the firmware from latest development instead of using the binary?

thanks a lot

Martenz, 2021-02-21:
By the way I have also the TBeam V.1.1 .. but I can't find the way to flash it.. it comes with softrf installed and has three buttons no switch to power on /off .. no way to erase or start the flashing.. it goes in sort of timeout.. any idea? tried without battery, holding the power button or the reset button.. same thing.. but I can inspect the serial on the COM port so for sure I'm connecting.. thanks

danielvelezbravo, 2021-02-23:
I have flashed all the 1.1  three button ttgo without any issues.  Are you
following all the indications on the manual?  Try to look up for the pdf
manual under doc on github and fillow every step.  Use the links on the
manual to get the flashing software.

Martenz, 2021-02-23:
> I have flashed all the 1.1 three button ttgo without any issues. Are you following all the indications on the manual? Try to look up for the pdf manual under doc on github and fillow every step. Use [...]

Yes I’m following the doc.. I tried also removing the battery.. same error with esp32 dev flashing tool and with arduino ide or platformio .. maybe it depends on oled and bmp280 that I soldered directly to the board? I found also some other reference saying I should ground pin 0 to ground with 1 to 10 K resistor.. will try like that .. maybe my board has some problems, should receive a new one soon and test there the flashing process before adding oled and bmp280. Let you know. Thanks

Martenz, 2021-02-23:
> GxAircom is already able to transmit and receive Fanet, so you will see and you will be seen by other Fanet-devices like Skytraxx.
> It is also able to transmit legacy-protocol like softrf. But only [...]

Can I ask you why flarm is not received? Which are the technical difficulty of receiving and processing also flarm? 
in my place for instance there are a lot of gliders using flarm, it is already amazing they can see me with this, would be nice if I can also see them plus all fanet devices around.

thanks for your time.

martino

danielvelezbravo, 2021-02-24:
I have ones with oled and other without and even the version 0.7 with
switch... the flashing process is the same, even for the heltec.

danielvelezbravo, 2021-02-24:
Flarm uses a licensed protocol with encription and maybe (not sure) it
needs a specia propietary module to do so. Every flarm device is registered
as unique.

Martenz, 2021-02-24:
> Flarm uses a licensed protocol with encription and maybe (not sure) it needs a specia propietary module to do so. Every flarm device is registered as unique. El mar., 23 de febrero de 2021 17:47, Tenz <notifications@github.com> escribió:
> […](#)

I know but so softrf how can it works then? do you think they have some license to receive flarm?

by the way this is the error I get on serial .. I tried also the resistor on 0 pin grounded.. same error:

rst:0x10 (RTCWDT_RTC_RESET),boot:0xb (HSPI_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
ets Jul 29 2019 12:21:46

thanks a lot

gereic, 2021-02-24:
Let me explain:
Fanet and Legacy have different Lora-Settings for sending/receiving.
So for sending of Legacy-packages, we switch the Lora-chip to the settings for sending Legacy-package. When sending is finished the Lora-Chip is switched back to Fanet-settings and waits for receiving a package.
So you can set the Lora-chip only to one of both protocols.
For sending, you can switch before sending to the mode you need.
But you can't receive on 2 different modes.

There are some options:
1.) listen always to FANET and sending FANET and Legacy (this is the way we do now).
2.) listen always to Legacy and sending FANET and Legacy.
3.) switch every (let's say 2sec.) between FANET and Legacy-receiving and sending FANET and Legacy.
     In this case you can only receive FANET or Legacy for a proper time, when it is active.

Hope this explanation is ok for you.

nepat, 2021-02-24:
If the software in the future were to support two LoRa radio modules for parallel Flarm and Fanet reception, then I could integrate that into upcoming versions of the PCB for the outdoor basestation. I would just have to switch to a different more compact format of the LoRa receiver. With the current (Lilygo hat) receiver the protruding SMA socket eats up too much build height in the case for fitting two receivers in the available space.

Martenz, 2021-02-25:
> Let me explain:
> Fanet and Legacy have different Lora-Settings for sending/receiving.
> So for sending of Legacy-packages, we switch the Lora-chip to the settings for sending Legacy-package. When [...]

Super thanks now it is more clear, as far as I'm not a super developer can I help in some way let's say writing down some documentation, work on 3d cases or whatever ? 

Thanks a lot I see a very nice future for this project

Martino

Martenz, 2021-03-03:
Hi I finally got an othet TBeam and this works with Platformio and also I can flash it. Installed GxAir 5.4.1.

One question I'm trying to compile from Platformio now, but I did a mistake I soldered the BMP280 on PINS VCC GN 2 13 instead of 13 and 14 .. where can I set/change the pins for BMP280 in the code? I'm not so good in soldering don't want to destroy the board.. if possible to set these PINS .. they are also in line so I can solder the baro directly close to the board.

thanks

Martenz, 2021-03-03:
Quick Update, after I soldered the bmp280 I realized where to edit the SCL and SDA pins in the code, but I see that BME280 library is used not bmp280.. might be  a problem to identify the sensor? 
I also got the same flashing problem now so I guess that with mp280 soldered on VCC 5v GN 13 and 2 there are some problems syncing the device. Tomorrow I will try to unsolder the sensor and see if that fix the flashing.

gereic, 2021-03-04:
Now it is clear, You have not used the standard-pinout for GXAircom.
Pin 2 is used when uploading a new program.
https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection
The lib is also able to be used with the bmp280.
I suggest, that you use the standard-pins which are in the readme.txt
Then all is working.

closed 2021-03-04 by gereic