gereic/GXAirCom #32: Wrong ID read from softrf Fanet

Martenz, 2021-02-22:
Don't know if it is a real issue or maybe a problem of softrf mark ii firmware but I get the wrong ID from all my devices set to fanet while receiving with GXAir.

Eg. device real ID 0D3724 read as 073724

[image]
[image]

danielvelezbravo, 2021-02-23:
You have a really old version of GXairCom.  Try to upgrade to version +4

Martenz, 2021-02-23:
> […](#)

Yep sorry that was old test, now I updated but I'm having the same problem with v.4.5.1 installed.

same ID one side: 3C003C and GxAirCom get 07003C ...

thanks

nepat, 2021-02-23:
That is actually all OK and fully intended. Older versions of GXAircom used a FANET Id with 07... Since a while GXAircom has its own FANET manufacturer ID and now it begins with 08....

Martenz, 2021-02-23:
> That is actually all OK and fully intended. Older versions of GXAircom used a FANET Id with 07... Since a while GXAircom has its own FANET manufacturer ID and now it begins with 08....

But my other two devices are softrf not gxaircom.. seems the first 3 values are not decoded properly.. maybe I’m missing some concepts.. thanks

closed 2021-03-04 by gereic