gereic/GXAirCom #35: LilyGo T-Watch board compatibility

Martenz, 2021-03-04:
Hi,

I just got one of these  LilyGo T-Watch including Lora, GPS, wifi and bluetooth:

https://it.aliexpress.com/item/33038999162.html?spm=a2g0o.productlist.0.0.654a12b27nXXyp&algo_pvid=4da35557-3c3f-4cd0-bcef-c2d20887209b&algo_expid=4da35557-3c3f-4cd0-bcef-c2d20887209b-1&btsid=2100bde116148632475175498e00ee&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

do you think it will be possible to configure also this board to have it working with GxAirCom?

I tested a flash of the firmware and the only thing that works of course is wifi and GxAir wifi interface.. the screen is a tft so I imagine it will need some recode for displaying data but to enable Lora and GPS I guess it can be done setting the correct pins..

thanks

danielvelezbravo, 2021-03-04:
looks like 433mhz lora module. Not FANET frequency (868-915mhz)

Martenz, 2021-03-04:
> looks like 433mhz lora module. Not FANET frequency (868-915mhz)
> […](#)

they have both, I received 868-915 mhz module

Martenz, 2021-03-30:
I was able to set the Lora to receive properly the fanet protocol.. now I get these kind of rx..

`Recvicer Done .. [size:45] -> radio_rx 4108142e8732417f3306c990007f00 -43 6

 0A 0D 3E 3E 20 72 61 64 69 6F 5F 72 78 20 34 31 30 38 31 34 32 65 38 37 33 32 34 31 37 66 33 33 30 36 63 39 39 30 30 30 37 66 30 30 20
rssi:-43 snr:6`

Which funciton or fanet object should I use to decode the data properly? or where should I then pass the string to then continue with the code? any hint?

thanks,

Martino.

Martenz, 2021-04-19:
Closing this because the TWatch 2019 is obsolete and newer versions don't have the lora module. Not worth investing time on this from my side.
Thanks anyway.

closed 2021-04-19 by Martenz