gereic/GXAirCom #40: NEO-M8N GPS compatible?

Martenz, 2021-03-12:
Hi I just received one of this:

TTGO T-Beam ESP32 868 MHz NEO-M8N GPS WiFi

it has a different gps module the Neo m8n, is it compatible or should I import a different library and compile source again?

thanks,

Martino

Martenz, 2021-03-12:
One note: 

it comes with softrf installed, I flashed the GxAirCom 4.5.1 and 4.6.0 but GPS not fixing yet..

thanks

gereic, 2021-03-12:
Go to settings and press configure GPS.
The esp will restart.
Then you should get a fix.
Is the red led near GPS blinking??

Martenz, 2021-03-12:
> Go to settings and press configure GPS.
> The esp will restart.
> Then you should get a fix.
> Is the red led near GPS blinking??

Ok so it should be compatible perfect! maybe it's my device..
The red led never blinked.. I flashed back the original softrf and after 15 minutes I got a fix .. then it worked properly.
Before I tired few times to press Configure GPS from GxAir settings but the device never restarted, when I press the button nothing happens.

I'm flashing back now the GxAir 4.6.0 .. I changed also the GPS antenna with a new one.

thanks, 
Martino.

nepat, 2021-03-12:
If any other software was installed on the device (SoftRF, etc.), then a memory erase should be performed with the ESP32 flasher software before flashing GXAirCom

Martenz, 2021-03-12:
> If any other software was installed on the device (SoftRF, etc.), then a memory erase should be performed with the ESP32 flasher software before flashing GXAirCom

yep I flashed with esp32 dev tool, erasing also before.

Martenz, 2021-03-12:
I've just tried also the 4.6.1 still no red blink.. will try to erase and build from platformio.

with platformio for T-Beam V1.1 shoud I use the [esp32_base] env plus the [env:psRam] env and comment out the others to compile quicker?

thanks 

Martino

gereic, 2021-03-13:
Can you check, if you see any nmea -output?
The red led is only the pps signal, when it is off, you don't have a gps-fix.
So set output to serial, check output nmea and look at the serial-terminal if you see nmea sentences.
If so, the software is working. You can try different gps-antenna.

nepat, 2021-03-13:
What type of TTGO T-Beam V1.1 do you have ? That board is also sold with the newer SX1262 LoRa Chip. That chip (the SX1262) to my knowledge is not supported by GXAirCom.

With the GPS Antenna that are supplied with the boards by Lilygo the time to first fix on the GPS can be extremely long and requires a very clear view of the entire sky. I have had good results with this small active antenna: https://www.aliexpress.com/item/32808970163.html?spm=a2g0s.9042311.0.0.27424c4dhagmwx

Martenz, 2021-03-13:
> What type of TTGO T-Beam V1.1 do you have ? That board is also sold with the newer SX1262 LoRa Chip. That chip (the SX1262) to my knowledge is not supported by GXAirCom.
>
> With the GPS Antenna that [...]

Got it working with different gps antenna and few minutes waiting in my garden.

Thanks. Now it fixes quite quickly 

Martino

closed 2021-03-13 by Martenz