gereic/GXAirCom #41: Volume not affected by second button

Martenz, 2021-03-17:
Hi I added both BME280 and Passive buzzer to TBeam v1.1,

everything is working except for the volume, should it change pressing the middle button? I see the icon change from low to high volume but seems my buzzer is always the same.

Moreover it would be nice to add also "no volume'' at all to disable the buzzer with the button also, so

OFF - Low - Mid - Hig cycle of the button

thanks

Martino

gereic, 2021-07-18:
I will double-check that

gereic, 2021-11-02:
Which amplifier do you use ?

Martenz, 2021-11-07:
I ordered these

ARCELI 2 Pezzi PAM8302 Scheda di Sviluppo Amplificatore di Potenza Audio Classe D da 2,5 W a Canale Singolo https://www.amazon.it/dp/B07RDJN2V1/ref=cm_sw_r_cp_api_glt_i_A2DSBTQV1BCH9Q208YXD?_encoding=UTF8&psc=1

closed 2021-12-26 by gereic