gereic/GXAirCom #42: eink 2.9 still not working for me

Martenz, 2021-03-18:
Following 
https://github.com/gereic/GXAirCom/issues/36

I desoldered the two resistors but still not working the display.. I tested the display and it works with arduino nano.

Should I remove or also do the bridge after removing the resitors?

any other idea why it doesn't work? exact same setup except I have also the BMP20 maybe?

Thanks,

Martino

_Originally posted by @Martenz in https://github.com/gereic/GXAirCom/issues/36#issuecomment-801764004_

MaUwJo, 2021-03-19:
Sorry - I can't help - maybe you ask in the  facebook group:
https://www.facebook.com/groups/1130097100726170
or try to build some eink examples and load them onto your device (use the same pins) - The debug messages are maybe better then.
- I did the same to test something with GPS: https://github.com/MaUwJo/test-ublox/blob/main/doc/test.txt

gereic, 2021-03-19:
[image]
Have you removed these resistor's??

Martenz, 2021-03-19:
Yes removed

Martenz, 2021-03-21:
I tried to run some GxEPD2 examples 

[example](https://github.com/ZinggJM/GxEPD2/tree/master/examples/GxEPD2_Example)

adding and setting the pins and the SW SPI as done in GxAir code:

```
#define EINK_BUSY     33
#define EINK_RST      4
#define EINK_DC       32
#define EINK_CS       15
#define EINK_CLK      0
#define EINK_DIN      2

GxEPD2_BW<GxEPD2_290, GxEPD2_290::HEIGHT> display(GxEPD2_290(EINK_CS, EINK_DC, EINK_RST, EINK_BUSY));
display.epd2.init(EINK_CLK, EINK_DIN, 0, true, false); // define or replace SW_SCK, SW_MOSI
display.init(0); // needed to init upper level
```

but still not working the eink with my tbeam v1.1 (neo6M) and **desoldered resistors**.

Moreover using DIN on 2 I need to unplug it before uploading the code with arduino ide otherwise the connection goes to timeout.

Will try to check again all pins if they are soldered properly otherwise I can try with an other board but I would like to know what removing the two resistors does.. is it still possible to use back the olde.. don't want to trash too may boards.

thanks,

Martino

Martenz, 2021-03-31:
Update

I checked connections and resistors to be completely removed but if you can I would like to see a picture of removed resistors to see if I did it properly.
I changed the pin for DIN in the code to 35 and built the latest (4.8.4) firmware and flashed to the board through platformio:

- now bmp280 is recognized and works and also the buzzer
- still eink not working

What can I try more?

thanks,

Martino

hehoe-ux, 2021-04-12:
I have encountered the same problem. It seems new displays (V2 - FPC-7519 rev.b on the display cable) are controlled differently. 
The solution is to copy the two files GXEPD2_290_t94.cpp and GXEPD2_290_t94.h from a current GxEPD2 lib from /src/epd/ into the corresponding lib directory in the project. Adjust includes and then use the lib in Screen.cpp:

GxEPD2_BW<GxEPD2_290_T94, GxEPD2_290_T94::HEIGHT> e_ink(GxEPD2_290_T94(EINK_CS, EINK_DC, EINK_RST, EINK_BUSY));

Quick and dirty but it works :-)

Martenz, 2021-04-12:
> I have encountered the same problem. It seems new displays (V2 - FPC-7519 rev.b on the display cable) are controlled differently.
> The solution is to copy the two files GXEPD2_290_t94.cpp and [...]

Thanks will give it a try, I also noticed that the 2.9 eink wave share v2 displays on Amazon have different controllers and bunch of different resistors in pictures but they don’t reference properly in description the version of the display so I guess this can be the problem.. I might got a different version than expected but I’m not so expert on this.

Thanks, meanwhile my tbeam board v1.1 with detached antenna and my weird display are on their way to Gerald so he can do also some more tests. Hope this can help also other people having same issue and fix it properly.

Martenz, 2021-04-19:
Closing the issue here, so with V2 you will need to compile with new libs (that sooner I guess will be included in the development compatibility) but for now stick to the V1 Eink which has a different controller and a lot more resistors on the connector side (on the  back) .. just to understand what you are buying because Amazon descriptions usually don't clarify which version you are buying. 

Thanks to all for the support now I have a fully working eink GXAir!

pixyfox, 2022-04-09:
Hello, do we still need the V1 version? They are at 2.1 now. 
Thanks