gereic/GXAirCom #43: Kobo

parahooners, 2021-03-21:
@gereic is there a kick start guide on the Kobo interface ? curious as I have one spare and would like to hook it up, happy to do a write up on this to help others if I could have a bit of direction on getting started,

Great work on the fuel meter, iv had a few stumbles with the sensor but slowly getting there now..

Matt

gereic, 2021-03-21:
There is no really kick start guide.
I can describe it just in a few words.
If you like, you can add it to the wiki.

- GND Battery to GND on T-Beam
- VCC Battery to Battery VCC on T-Beam
- TX on T-Beam to RX on Kobo
- VCC from Serial-interface on Kobo to VP (IO36) on T-Beam

Enable Pin36 in setting, so that T-Beam switches on and off with kobo

That's it.

If you like, you can add a Baro-sensor to.

parahooners, 2021-03-21:
@gereic thanks for the fast response !!

Is there a preferred software for the KOBO at this stage ?

gereic, 2021-03-21:
LK8000, cause it can receive FANET-MSG
So you have to setup a device to LK8EX1 and a second device to FANET
Both devices use the same port (/dev/ttymx0).
Baudrate 115200

parahooners, 2021-03-21:
@gereic Thanks for the info and great work on the TTGO TBEAM, had a quick flight this weekend and all worked well

closed 2021-03-21 by parahooners