gereic/GXAirCom #44: TTGO platform IO error

parahooners, 2021-03-25:
[image]
 
Hey, can any one assist getting this error after flashing

parahooners, 2021-03-25:
@gereic

gereic, 2021-03-25:
BMPM085 is not supported yet ?
Is this your own code ?

gereic, 2021-03-25:
I think you didn't downloaded the code.
Looks like a testprogram for a BMP085

parahooners, 2021-03-25:
https://user-images.githubusercontent.com/73525787/112540976-bbd22400-8daa-11eb-96a4-7b121d8ed809.mp4

Uploaded the code directly from the code link getting this ....

@gereic  not my own firmware straight from the github page

gereic, 2021-03-25:
This is not GXAircom, cause the size is 21.9% and that is much too less.
Don't know which software you try to compile, but this is not GxAircom.
You can download the software directly with ESP32 tools, that is easyer for beginners.

closed 2021-03-25 by gereic