gereic/GXAirCom #46: Add meaningful Readme

dwt, 2021-04-16:
Hi there,

I've heard some positive things about this project, but it's really hard to dig deeper as there seems to be no documentation about it's abilities?

I would have liked to see:
- Some information about what the software does, which hardware it requires and where to find further documentation in the Readme.md file
- A link to a website (if it exists) somewhere

I guess some of this already exists, as people seem to be able to use it - but this documentation is definitely lacking in the repository

brunotl, 2021-04-16:
it's not very hard to find usefull doc :  https://github.com/gereic/GXAirCom/tree/master/doc

dwt, 2021-04-16:
Am I missing something? I see quite some Protocol Specifications, 3D files for cases, but am missing:

- General capabilities of the project (which protocols are supported, what hardware is supported)
- What Smartphone Apps are supported via Bluetooth?
- Links to pages that document setup / first use?
- Link to documentation on how to build it
- Link to documentation on development roadmap / how to send patches

This document seems to be close: https://github.com/gereic/GXAirCom/blob/master/doc/20200723%20-%20GXAirCom%20-%20A%20LoRa%20communication%20device%20for%20free%20flying.pdf 

Is there a reason not to link that in the readme together with some more information?

nepat, 2021-04-16:
Github is targeted at the development community. That being said Github also includes the tools by which contributions can be made to a project.

Martenz, 2021-04-19:
You can find also some more info in the wiki

[https://github.com/gereic/GXAirCom/wiki](https://github.com/gereic/GXAirCom/wiki)

dwt, 2021-05-03:
@gereic just a thought - if you want the open source community to eventually contribute a useful readme that you don't want to write yourself, it would really help to keep the bug open and mark it as 'help wanted' or just leave it open with a note saying so.

dwt, 2021-05-03:
👍

dwt, 2021-05-06:
Is there some developer focused documentation somewhere? I.e. how you guys accept patches, what the review process is, how testing is done?

dwt, 2021-05-10:
I'd love some link in the readme to a help chat, perhaps something like a Gitter channel or a forum where users can turn to find other users and possibly get some support.

Does something like that exist somewhere?

closed 2021-12-26 by gereic