gereic/GXAirCom #47: Fuel meter window

parahooners, 2021-04-23:
@gereic 

Have been testing the fuel meter and the vibration from the engine is creating quite a lot of noise on the sesnor.. 

Iv written a small section of window sampling code which works well to filter out the noise, is this something that could be implement... to the voltage spikes are smoothes out giving a average reading....

parahooners, 2021-04-23:
https://drive.google.com/file/d/1iZXwJqpWXt3we06u-MLNOpKjUkAN4XNZ/view?usp=drivesdk

gereic, 2021-05-02:
Can you try v4.7.1 from dev-branch ?
https://github.com/gereic/GXAirCom/tree/dev/bin/dev
I don't have a sensor here, so it is not possible to test.
Please give me feedback, if it works.

parahooners, 2021-05-18:
@gereic  have tested with new fw and the readings are very much improved, this really now is a hardware limitation the vibrations caused bt the tank send the sensor crazy... hope full the weather will improve and I can stretch the sky legs...

closed 2021-05-18 by parahooners