gereic/GXAirCom #48: GPS coordinates are not updating in GS mode

bortek, 2021-04-25:
Testing GroundStation mode on T-beam v1.1.  

According to the PDF doc the coordinates should be automatically set using GPS if they are set to zeroes in the settings. But this does not seemto work. When checking i the INFO page after startup the coords are still zeroes.

Another issue is that the GS is not visible on the OGN network but the test aircraft is showing up. I can see other GS on the map but not mine.

bortek, 2021-04-25:
[image]

gereic, 2021-04-26:
When you set GXAircom to Ground-Station, you have to set GPS-coordinates in the settings-page.
It will not automatically used from GPS.

closed 2021-04-26 by gereic