gereic/GXAirCom #53: Version 4.8.0_UTC Time wrong

ParaJesper, 2021-05-10:
Flashed my TTGO (V1.1) unit to FW 4.8.0. today. Reading through the info screen I found the UTC Time reverted to 1970-01-01. At first I thought this would be updated following GPS-fix, however after getting a good fix it remained at this date.

Previous software was erased before flashing and unit was reset and restarted before starting this issue.

2nd, minor issue is the functionality of the menu option, "Basic, Advanced and Expert". This does not show anything different in the menu layout??

[image]

danielvelezbravo, 2021-05-10:
I think that when you have another sensors connected to the module, the
differences on the menu options will pop up. But if you don´t have a
pressure sensor (as an example) then you will not have the audio, vertical
speed configuration options available.

ParaJesper, 2021-05-10:
> I think that when you have another sensors connected to the module, the differences on the menu options will pop up. But if you don´t have a pressure sensor (as an example) then you will not have the audio, vertical speed configuration options available.
> […](#)

Ah,that could very well be the case. Thnx!!

closed 2021-05-18 by gereic