gereic/GXAirCom #55: baro ms5611 with GY-63

erwin4u-alt, 2021-05-17:
Hello, I've bought a new Baro board GY-63 (ms5611) and can see the appropriate section in the web console (Vario) but without values. In the documentation I figured out, that only the GY-86 was implemented. Has the GY-63 other specifications or is my module corrupt?

gereic, 2021-05-17:
The GY-63 has only the MS5611 on Board.
The GY-86 has the MS5611 (Baro) and a MPU6050 (accelerometer, gyroscope).
So, it is the wrong board.

closed 2021-05-17 by gereic

erwin4u-alt, 2021-07-27:
Hello,
Could you assist me again please?
I’ve bought an GY-86 but I don’t receive barometric data. I’m using the TTGO T-Beam T22_V1.1 and the latest version of GXAIRCOM (v4.10.1). The bar board is connected to Pin 13,14. If I connect the board, I get the related menu in the setting page, but no data for altitude and temperature…

Do I need an additional configuration or something else??

Cheers - Thomas

gereic, 2021-07-27:
Can you try it with v4.11.0 please ?

erwin4u-alt, 2021-07-27:
Unfortunately the same

gereic, 2021-07-29:
Can you post the output of the serial-console ?
You should something like
[I][Baro.cpp:416] initMS5611(): found sensor MS5611
[I][Baro.cpp:430] initMS5611(): MPU6050 connection successful
[I][Baro.cpp:454] initMS5611(): tempcorrection: t0=20.00 z0=0.00 t1=0.00 z1=0.00
[I][Baro.cpp:456] initMS5611(): Initializing DMP...

You get the GPS-Altitude only from GPS.
If you have connected a E-Ink, it shows the Baro-Alt in case you don't have a GPS-Fix.
In the INFO-Page, you should see the temperature.

[image]

erwin4u-alt, 2021-07-29:
Hello,
Something is wrong with the MPU6050.
What do you think - is my GY-86 defect?

[I][main.cpp:4557] taskEInk(): stop task
[I][Baro.cpp:416] initMS5611(): found sensor MS5611
[I][Baro.cpp:432] initMS5611(): MPU6050 connection failed
[E][Preferences.cpp:493] getBytesLength(): nvs_get_blob len fail: axScale NOT_FOUND
[E][Preferences.cpp:493] getBytesLength(): nvs_get_blob len fail: ayScale NOT_FOUND
[E][Preferences.cpp:493] getBytesLength(): nvs_get_blob len fail: azScale NOT_FOUND
[I][Baro.cpp:454] initMS5611(): tempcorrection: t0=20.00 z0=0.00 t1=0.00 z1=0.00
[I][Baro.cpp:456] initMS5611(): Initializing DMP...
[I][Baro.cpp:464] initMS5611(): gyro not calibrated
[I][Baro.cpp:474] initMS5611(): acc offsets ax_offset=-1529,ay_offset=-5120,az_offset=2
[I][Baro.cpp:489] initMS5611(): scale ax_scale=1.00,ay_scale=1.00,az_scale=1.00,
[I][Baro.cpp:490] initMS5611(): acc offsets ax_offset=0,ay_offset=0,az_offset=0
[I][Baro.cpp:491] initMS5611(): gyro offsets gx_offset=0,gy_offset=0,gz_offset=0
[I][fmac.cpp:1144] readAddr(): dev_id=08E538

gereic, 2021-07-29:
Can you send a picture of your board.
Where have you ordered it ?

erwin4u-alt, 2021-07-29:
I bought this: https://www.amazon.de/gp/product/B06Y2BKB5H/ref=ppx_yo_dt_b_asin_title_o06_s00?ie=UTF8&psc=1 <https://www.amazon.de/gp/product/B06Y2BKB5H/ref=ppx_yo_dt_b_asin_title_o06_s00?ie=UTF8&psc=1>