gereic/GXAirCom #57: Heltec LoRa Module - Battery Status

AHausBy, 2021-05-21:
When attaching a full loaded 3.7 V 1800mAh Lithium Polymer Battery to the Heltec LoRa Modul, The Battery Status is shown as empty on the OLED Display.

gereic, 2021-05-26:
Have you connected the resistors to pin 34 as described in the readme ?

******** BATTERY-Voltate *******
                       ____           ____
BATT_VOLT_PIN 34   ---|____|----|----|____|-----
                  GND  100K   GPIO34   27K     Batt +

AHausBy, 2021-05-26:
Did not realize that TTGO Lora32 is the Heltec Board.
That means i had to add 100k Resistor connectet to GND and GPI34* and 27K Resitor to Vext and GPI34* on my Heltec V2 Board, is that correct ?

gereic, 2021-05-26:
Yess that is correct.
In settings, the Heltec and TTGO-Board are the same.
In the /bin-folder, there is a readme, where you can see the exact wiring.

AHausBy, 2021-05-26:
Ok, i will try that, thank you

closed 2021-05-26 by AHausBy