gereic/GXAirCom #58: Invalid data in APRS stream

Meisterschueler, 2021-06-05:
In the APRS stream of the OGN we often see complete invalid messages from GXAirCom devices (wind from 2147483647 degree...):

```
FNTFBACA4>OGNFNT,qAS,Sonthofen:/084427h5807.34S/00232.69W_2147483647/2147483647g2147483647 0.0dB
FNT060001>OGNFNT,qAS,Huenenb2:/084351h5807.34S/16053.67E_2147483647/2147483647g2147483647 0.0dB
FNT060001>OGNFNT,qAS,HuenenWX:/084435h5807.34S/00000.23E_2147483647/2147483647g2147483647 0.0dB
```

But also these messages are invalid:
`FNT085890>OGNFNT,qAS,WyPgvNw:/084545h4756.32N/01444.20E_359/004g006t070r4310p30830h54b10176 0.0dB`

We see r4310 (=1094mm rain in the last hour), but the rainfall of the last hour is defined as 'rxxx' so we have just 3 digits. Same for p30830.

All these beacons are ignored by the python-ogn-client library. Please check out the APRS specification: http://www.aprs.org/doc/APRS101.PDF

gereic, 2021-06-05:
Thanks for your response,
I think I know already, where the error is.
It seams, that a FANET-Packet was interpretet as Weatherdata.
I will do a validation in Code.

gereic, 2021-06-12:
I have changed the code.
Can you check, if "WyPgvNw" is still sending invalid data?
Is it possible, that I can check this either?

Meisterschueler, 2021-06-12:
You could check it by your own with https://github.com/glidernet/python-ogn-client and this script:
https://pastebin.com/rmmKPaZc

closed 2021-07-18 by gereic