gereic/GXAirCom #60: Trouble getting GPS signal

landorg, 2021-07-04:
Hi,
first thank you very much for that great project.

I'm having trouble getting a GPS signal with my new TTGO Meshtastic T-Beam V1.1 ESP32.
I flashed the most recent software and made the recommended settings. Everything looks good so far except it won't get a GPS signal. Display shows NO GPS-FIX and web ui shows all zeros in the gps status. 

logs from serial monitor:
https://gist.github.com/landorg/4d21363e9cf004ec99cdcccf31236bff

Any idea what could be wrong here or tips to debug?
Thanks

gereic, 2021-07-05:
Hi,
It looks like you have a bad antenna.
Try to get a GPS-Fix in a free area.
You should be outside and nothing around in a distance  ~10m.
When the red led is flashing, the GPS is fixed.
Or you can try with a different GPS-Antenna 
[https://de.aliexpress.com/item/32972306714.html?spm=a2g0o.productlist.0.0.281772f4MXENps&algo_pvid=d783a773-41c7-4aa7-8419-7f250d0d4825&algo_exp_id=d783a773-41c7-4aa7-8419-7f250d0d4825-51](url)

Gandalfderweisze, 2021-07-18:
Hi, great project!

I had also troubles to get a GPS-fix. Using USB-Power, no battery. There was only a blue LED steady shining behind the OLED. No red led was flashing.

Hardware: 
LILYGO®TTGO T-Beam V 1,1 ESP32 868 Mhz WiFi Bluetooth Mod
ul ESP32 GPS NEO-6M SMA 18650 Batterie Halter Mit OLED

Print at the board:
T22_V1.1
20210222.

GPS      ESP32
TX        IO34(RX)
RX        IO12(TX)

GXAircom v1.5.3 flashed according instructions in the manual Rev. 20200728 GXAirCom 1.2.2

Antenna looked fine, at a first try I was outside for 10 minutes, no sucess. So I already gave up. However then I was thinking about my very old Garmin GPS12 which needed about 15minutes to get a fix from a cold start. This time is needed because tables has to be downloaded from the satellites at a very low bit-rate. So I gave another last try. I set myself a timelimit about 1 hour. After 20 minutes I had sucess, now 5 to 7 satellites are in view. So 10minutes are maybe to short after a cold start for this version of board. Try 1 hour or so....

greetings
Jürgen

gereic, 2021-07-18:
Hi Jürgen,
Please use the latest version V4.10.1 from the bin-directory.
Then go to settings and press configure GPS.
Board will setup GPS-configuratio and then reboot.
After that place the board to area, where you can see the sky.
Recommend is that there are no obstacles around 10m.
When the red led near GPS is blinking every second, the GPS is ok.
The have a look in the info-page,if you have a gps-fix.
If the red led doesn't start blinking after 30min. You have to install a better GPS antenna.

Gandalfderweisze, 2021-07-18:
Hello, 

updating via cell phone doesn't work. The device never rebooted. Updating via flashtool was possible, but only after a reset. Works fine now, GPS-fix is much faster (within 1-2 minutes or so) and there are much more options to use. 

Thank's a lot! 

Und wenn Du mal in die Steiermark zum Kulm kommst melde Dich, hast ein paar Bier gut ;-)

greetings
Jürgen

closed 2021-07-19 by gereic