gereic/GXAirCom #63: No transmission or receiving data on 4.10.1

Skyhighpara, 2021-07-25:
Hardware: LILYGO TTGO T-Beam V1.1 with SX1262 chip

Flashed 4.10.1 and tried all receive and transmit RF modes in both, air and ground station mode. 
Not a single package received (with active FANET+ transmitter next to it; so both Fanet and Flarm transmitted).
All counters at 0.

Downgraded to 4.9., immediately received and transmitted (once GPS was fixed) for FANET.
No Legacy/Flarm transmission or receive still, on any setting or combination of RF-Mode. 

Back to 4.10.1 (both psRAM versions), again, nothing on identical settings. 
I dont know what changed, but it seems to have broken any transmission with the SX1262 chip.
Also, LEGACY mode seems to not work on this specific hardware?

gereic, 2021-07-25:
Can you try to update to 4.11.0 from dev-branch.
Update Firmware and spiffs please.

Skyhighpara, 2021-07-26:
Thanks for the quick solution!
4.11. seems to be fully working:

- receiving on both FANET and LEGACY/FLARM in both modes, AIR and GROUND.
- Transmission is working for both RF modes in AIR and for FANET in GROUND station mode - no LEGACY TX here, I assume that is intended.

As a small side note; 
the FANET+ module from Skytraxx is transmitting both, FANET and FLARM, however only FANET is being received by my module. At the same time, planes, helicopters, gliders etc. FLARM is received - LEGACY mode seems to be working.
I cannot see if there is a filtering getting rid of the duplicate signal or some protocol issue.

danielvelezbravo, 2021-07-26:
Are you sure that Fanet+ on skytraxx transmits FLARM?  I think i read
somewhere that FANET+ means it can transmit FANET and receive FANET and
FLARM, but it does not transmit FLARM "per se".

Skyhighpara, 2021-07-26:
Not quite; It does transmit and receive FANET and **transmit** FLARM, but cannot **receive** it. 100% sure.
1. Flarm transmission is necessary to be visible to copters, planes and gliders - Us paragliders seeing them does not help to much as we are too slow to dodge anyway.
2. You can see that Skytraxx module on OGN, you will often see people appearing twice, with both their Fanet and Flarm ID from the same device.
In Air mode I believe the counter going up with LEGACY is from my device, however in received aircrafts only the FANET part shows up.

danielvelezbravo, 2021-07-26:
Ah. Understood.

gereic, 2021-07-26:
You are completely right.
Fanet+ means send and receive Fanet and send also Flarm.
You should see, that the Legacy-Rx-counter increases.
I think Skytraxx sends every second Flarm on 868.4MHz.
On the received aircraft-page, you see how the aircraft was received last.
So for the Skytraxx, it will flicker from Faber to Legacy from time to time.

Skyhighpara, 2021-07-28:
> You are completely right.
> Fanet+ means send and receive Fanet and send also Flarm.
> You should see, that the Legacy-Rx-counter increases.
> I think Skytraxx sends every second Flarm on 868.4MHz.
> On the [...]

Thats the behaviour I have been expecting. However I only see the FANET count going up, nothing in LEGACY mode for my device.
I have to add that I am using the Strato from Skybean, which is also in very early development still; it is however using the Skytraxx Fanet+ module.
I have seen my device registered with its Flarm and Fanet ID both to OGN when it was picked up by another groundstation. 
For now, I can only report that this specific combination seems to have trouble with FLARM/Legacy mode, but dont know which side the problem stems from.
Anyway, thank you again for the quick replies, I will report back once I learn more.

closed 2021-07-28 by Skyhighpara