gereic/GXAirCom #68: No GPS

parahooners, 2021-08-01:
TTGO TBEAM

Have tested GPS via Tinygps and confirm GPS module is working

no read out in info tab 

[image]

parahooners, 2021-08-01:
[image]

gereic, 2021-08-02:
Which Board do you have ?

parahooners, 2021-08-02:
It's the ttgo tbeam with 3 buttons 1.1 I think 

Have tried changing the settings tab

parahooners, 2021-08-02:
@gereic 

when trying to use the config this is the serial error im getting - the pin mapping for the board is correct for the GPS

[E][main.cpp:3758] setupUbloxConfig(): ublox: GPS not connected

parahooners, 2021-08-02:
main.cpp

have changed the settings here 

[image]

then to 

[image]

is this right 

[image]

still not working

parahooners, 2021-08-04:
Had a new module delivered today have tested on 3 different ones same results - AXP192 chip is picked up but the same error occurring

parahooners, 2021-08-05:
i can confirm that 4.8 works @gereic 

[image]

gereic, 2021-08-05:
Can you post a picture of the board?
Where I can see the version and date.
I have v4.11.2 on some boards running without any problems.
Please post also the settings-page (which board selected)

parahooners, 2021-08-06:
I got this one...

https://www.banggood.com/LILYGO-TTGO-T-Beam-ESP32-433-or-868-or-915Mhz-v0_7-WiFi-Wireless-bluetooth-Module-ESP-32-GPS-NEO-6M-SMA-LORA-32-18650-Battery-Holder-p-1320390.html?utm_source=googleshopping&utm_medium=cpc_organic&gmcCountry=GB&utm_content=minha&utm_campaign=minha-gb-en-pc&currency=GBP&cur_warehouse=UK&createTmp=1&ID=510804&utm_source=googleshopping&utm_medium=cpc_bgs&utm_content=sxxx&utm_campaign=ssc-gb-all-enw-0624-20lp&ad_id=444236244939&gclid=EAIaIQobChMI6c2gmYOd8gIVCbLtCh1JswUAEAQYBCABEgLFu_D_BwE

gereic, 2021-08-06:
This is the V0.7, so you have to select the right Board from the list.
This Board doesn't have a axp192

parahooners, 2021-08-06:
Yeh,

This is what's odd tho it says on the paper work it has the chip 😂

I will get some v1 boards I cheaped out on the boards

parahooners, 2021-08-06:
odd, so i flashed it via the firmware upgrade tab after flashing 4.8 and it blooming works.... i have no idea hahahah

closed 2021-08-06 by parahooners

parahooners, 2021-08-06:
Manged to fix one board, unfortunately the ttgo does not have boot loader pin , shorted 3v3 and I0 to force boot loaded erased using the tool and replaced and all worked.