gereic/GXAirCom #76: Feature Request: OpenSky Network Feeding

ketilmo, 2021-10-30:
Hi,

Thanks for creating a great FLARM/FANET+ solution. For the ground station, would it be possible to implement an option for feeding data to OpenSky Network? 

The following page has some more information about the technical details: https://opensky-network.org/forum/feature-updates/363-new-feeding-flarm-data-to-opensky

OpenSky Network is a EU-funded project that stores ADS-B and FLARM data for research purposed. The data can be accessed through and API. In addition, OpenSky Network features a live map of all current flights.

Kind regards,
Ketil

gereic, 2021-11-02:
Hi,
I can integrate that, but I need the Information, on how to feed the data to the opensky-Server
Servername/IP
Port
How to send data and which data to send.

Can you check, where I get this Information please.

ketilmo, 2021-11-07:
Hi @gereic,

That's great news, thanks! I'll get in touch with the OpenSky Network team and get back to you.

ketilmo, 2022-01-08:
Hi @gereic,

I have just heard back from the OpenSky Network team. This is what they write:

_If they are using ogn-decode within their software it will be straight away, as they will only need to add this line to the configuration file Demodulator:` { MergeServer = "flarm-collector.opensky-network.org:20002"; } ;`_

_Else, it will require some more time as our server accepts UDP messages in a very specific format as it is provided by ogn-decode. As this is closed-source we don’t have the code either but just what the C Classes used to prepare the UDP packets look like. It will take some time to extrapolate the information and also possibly to adapt the GxAirCom._

How does this sound?

closed 2025-02-19 by gereic