gereic/GXAirCom #78: BM280 baro sensor

petr0808, 2021-11-04:
Hi,

I'm trying to wake-up a TTGO V1.1 with baro sensor attached to pins 21 22, GND and 3V3. The firmware does not recognise the sensor at all. Not even when selecting unknown board to force HW detection. In this case the firmware detects the proper board version and resets to V1.1. If I test a similar setup but with v0.7 board, in case of unknown setting the baro sensor is detected and the vario submenu is available. In case of selecting of v0.7 version the sensor is again not recognized.
Is there a workaround or any solution for this issue? The HW setup is quite fixed due to the fact that the sensor is soldered to an "adapter board" attached to my TTGO directly.

gereic, 2021-11-04:
Hi petr,
As written in the readme, the baro has to be connected to 
******** BARO *********
SDA 13
SCL 14

petr0808, 2021-11-04:
Hi gereic,

Is there any chance to change that firmware wise? I should not redesign and reproduce the whole PCB on which the baro sensor is placed.

gereic, 2021-11-04:
It is not so easy, because the baro-task is seperated from the rest 21/22 is used for O-LED and axp192.
Do have already ordered the PCB's ?

petr0808, 2021-11-05:
Yes, the first proto is already ready since I've started with softrf. After the planned tests I''ve realized the existence of GXAircom that fits more to my porpose due to FANET and Flarm operation "in the same time". So therefore the easiest solution for my issue were the firmware update if it is possible without high efforts. By the way O-LED and axp192 will not be applied to the final system.

closed 2021-11-17 by petr0808