gereic/GXAirCom #82: GPS is not working with latest Dev Version

parapenT1sta, 2021-12-04:
TTGO T-Beam T22_V1.0 / T22_V1.1

Had to roll back to master version to make it working again. With Dev version when press config GPS at the end. Checked with LK8000 terminal and verified no GPS output was being sent. 

At the moment I'm with Master version but I can flash the Dev for version if some tests are needed.

gereic, 2021-12-06:
Can you try it with new Dev-Version please ?

parapenT1sta, 2021-12-06:
Same issue.

With Stable version:
[image]

With Latest Dev
[image]
:

gereic, 2021-12-06:
[image]
 I have checked that and it works.
Is the Board connected via bluetooth or serial ?

parapenT1sta, 2021-12-06:
When I press Config GPS it fails.
[image]

[image]

gereic, 2021-12-06:
Please download again the firmware directly from git and try again.
Maybe you have an old dev-version.
Make a screenshot from info-screen.
The compile-date should be from today.

parapenT1sta, 2021-12-06:
I didn't update after your message because I already had the latest version, this is the firmware I am using:

[image]

parapenT1sta, 2021-12-06:
Just ignore my "issue", if it is working for you I might be doing something wrong.
I will keep with 4.12 master version that is working perfectly fine for me.

closed 2021-12-06 by gereic

parapenT1sta, 2021-12-10:
@gereic I just wanted to leave my latest feedback, I went back on Dev releases to find which one worked for me, and I found the latest DEV 0.13 that works for me! 🥇
This build works perfectly fine with GPS, all releases after I only get the $PFLAU sentence: https://github.com/gereic/GXAirCom/commit/a0223933aea7dcdb497144d7431df5765e0b6f79#diff-97c8e95a046a07cbeb99e61fb3ae6d9665d265955d6fab5206f4c52e8180cdfb

parapenT1sta, 2021-12-11:
Found the issue, before was working even with the wrong board selected. Now with the changes the correct board must be selected. Sorry !!

gereic, 2021-12-11:
That is good to hear.
Thanks for your response

parapenT1sta, 2021-12-11:
@gereic Thank you for the support.