gereic/GXAirCom #84: Version 4.13 double definition of GPIO34

wielandsteffan, 2021-12-20:
I have a Heltec Board for flying. After installing V4.13 there ist no GPS data anymore. In Serial output GPSRX changed from GPIO12 to GPIO34 in the new Version. But i found that there is a double definition of GPIO34 in main.cpp   

PinGPSRX = 34;
PinADCVoltage = 34;

best regards Wieland

parapenT1sta, 2021-12-23:
I had a similar issue regarding GPS not working but I found out in my case that before I was able to use it even when the wrong board was selected but it seems that is not your case. https://github.com/gereic/GXAirCom/issues/82

Maybe @gereic can assist you better.

wielandsteffan, 2021-12-23:
When i change definition of GPSRX to GPIO12 in main.cpp and recompile V4.13 then it works as before.

parapenT1sta, 2021-12-23:
Submit the change here on Github so can be reviewed by @gereic

gereic, 2021-12-25:
@wielandsteffan Can you try latest dev-Version please ?

closed 2021-12-26 by gereic