gereic/GXAirCom #85: clone in VSCode with PlatformIo - errors

mm16bit, 2022-01-07:
Hi, thank you for your efforts! I flashed the ready binaries from the .bin folder and everything works great.

However, I have also cloned your last project in VSCode with PlatformIo. Unfortunately, I get a lot of errors and I don't know why. (see pictures). Do you have any idea where the problem is?

Kind Regards

[image]
[image]

vindolin, 2022-02-22:
> 
I've noticed that your info bar is missing the selected platformio environment, or did you hide yours?
Here's how it should look: [image]

gereic, 2022-02-22:
Looks like you have a problem with the include.
The BEEP_VELOCITY_DEFAULT_SINKING_THRESHOLD is in the beeper.h
So it doesn't find that file.
In the program the beeper.h should be included like this.
#include <beeper.h>

mm16bit, 2022-02-28:
@gereic It is the original project cloned from github into VSCode.
I just copied it again from Github to VSCode.No changes made at all.
As soon as I click on main.cpp I get about 40 errors (identifier xyz are not defined). The beeper.h is included (see picture).
[image]

matl-gh, 2022-03-04:
@mm16bit are these build errors or linter errors? I've got the latter as well, but build runs fine.

I guess the reason is that the includes are not the way it is normally done, but how platformIO expects it. For including own code default quotation marks (#include "xxx") are normally used, see https://gcc.gnu.org/onlinedocs/cpp/Include-Syntax.html. When compiling with platformIO, the lib folder is added to the default search folder and the files are found even without specifying the explicit path, your linter might not do this.

mm16bit, 2022-03-10:
@matl-gh it seems that the build is working. Everything is compiled without errors. I get a lot of warnings, e.g. that variables are "defined but not used".
I'm just trying to get back into C++ a bit and understanding code from interesting projects is very helpful.
I don't have any experience with platformio and VSCode, so the many errors and warnings are a bit confusing for me.

gereic, 2022-03-10:
What was the problem ??

mm16bit, 2022-03-10:
@gereic What do you mean? 
My problems are above and as I wrote : "_I don't have any experience with platformio and VSCode, so the many errors and warnings are a bit confusing for me._"
I would just like to understand where the errors come from

brunotl, 2022-03-10:
@mm16bit : to learn how to use VScode, the best start point is the VScode documentation ;-)

for your problem, you can start here : https://code.visualstudio.com/docs/cpp/configure-intellisense-crosscompilation

mm16bit, 2022-03-10:
@brunotl Thanks for the information, I had already looked at that and the include paths are all entered in the "c_cpp_properties.json".

closed 2022-12-27 by gereic