gereic/GXAirCom #87: Add description and project overview

dwt, 2022-02-08:
Currently the project doesn't provide a description set in Github., which makes it needlessly hard to find.

I propose a description similar to the one in the readme, or if needed a shorter one that at least mentions lora and fanet.

I would also like to suggest setting the respective tags.

What do you think?

dwt, 2022-02-22:
So for example searching GitHub for [aviation proximity awareness system](https://github.com/search?q=aviation+proximity+awareness+system&type=Everything) doesn't find the project, as well as searching for [aviation lora paragliding](https://github.com/search?q=aviation+lora+paragliding&type=Everything).

I would like to at least get some tags going: fanet, lora, paragliding, gliding, flarm, ogn

Maybe add tags for the supported boards as well.

For the description I would recommend something similar to what SoftRF uses, or what the readme states. Maybe something like this?

🪂 Multifunctional, compatible DIY aviation proximity awareness and messaging system with FANET+, FLARM and OGN support.

dwt, 2022-02-22:
Some more queries that don't find the project right now: [paragliding fanet](https://github.com/search?q=paragliding+fanet), [lora fanet](https://github.com/search?q=lora+fanet), [fanet](https://github.com/search?q=fanet), [fanet flarm](https://github.com/search?q=fanet+flarm).

I think the project should aim to be found by all of these at least.

gereic, 2022-02-22:
ok. Do you know how this can be done ?

dwt, 2022-03-11:
@gereic Do you need any additional info for you to add these tags and description?

gereic, 2022-03-11:
Changed things, you suggested.
Can you try a search ?

dwt, 2022-03-11:
All the searches I linked to above now find this project. Nice!

closed 2022-03-11 by dwt