gereic/GXAirCom #90: Feature request: T-Echo NRF52840 LoRa SX1262 868MHz support

herriett, 2022-02-20:
I have a bit the feeling that the momentum is shifting towards these newer boards. It is certainly a big advantage, that one can buy them completely ready to use. 

Are there plans to extend the support to this hardware in the future?

gereic, 2022-02-22:
Sorry, for now, there will be no support for that.
The reasons are:
1. case is not good (antenna plug can be turned)
2. no wifi-connection (so I have to build a bluetooth-app for settings
3. no connection (or not easy to solder) for speaker.

This is not an absolute no-go, but for the moment, there is no support planned.

lhaeger, 2022-05-03:
> Sorry, for now, there will be no support for that. The reasons are:
> 
>     1. case is not good (antenna plug can be turned)

Not sure which case you looked at, but the current version is really well made, not 3D printed anymore. 

>     2. no wifi-connection (so I have to build a bluetooth-app for settings

The T-Echo can be connected to Windows amchines as a storage device (double click reset immediately after USB connect), so a simple text file may be an easy alternative

>     3. no connection (or not easy to solder) for speaker.

While this may render the T-Echo useless as an audio vario, it would still make a great BlueTooth vario for use with XCSoar/XCtrack/LK8000/FlySkyHy etc.

>     4. no connection (not easy connection) for external devices like accelerometer, SD-card, ...)

This may change, there's already a microphone daughter board available, others should follow.
> 
> This is not an absolute no-go, but for the moment, there is no support planned.

Would you mind outline the high-level steps required to port GXAirCom with the above limitations to the T-Echo? I might give it a try then...

Until then, I work on improving SofRF's vario capabilities, since it already runs on the T-Echo and gives me a FLARM beacon (as a paraglider I do not need to see others, but I want to be seen by gliders and helicopters). It's just not that great as a vario (yet).

closed 2022-12-27 by gereic