gereic/GXAirCom #96: Feature Request: Send FNNGB-sentence with FANET-Infos in NMEA-Data-Stream

mwesterm, 2022-04-01:
GXAircom sends now all infos about received aircrafts in a $PFLAA-sentence.
To use all features in XC Guide for example it would be great if via FANET-received aircrafts generate a FNNGB-sentence.
It should be able to select, if infos via FANET received aircrafts should be forwarded as FLARM ($PFLAA) or FANET ($FNNGB) record.

gereic, 2022-04-01:
Is there already a official documentation about FANET-sentences ?

gereic, 2022-04-01:
GXAircom sends all FANET-Messages as FNF-sentence already.
XC-Guide is already receiving all messages, this is the reason, why XC-Guide can show weather-stations ether.

mwesterm, 2022-04-01:
You are right, XC-Guide also accept th FNF-Sequences and show aircrafts as received by FANET (and the FANET-Filter also works).

closed 2022-04-01 by mwesterm