gereic/GXAirCom #102: gy-86 "retired"

bzed, 2022-07-16:
Hi,

first companies seem to stop the production of the gy-86 module, MPU 6050 should also not be used for new projects anymore.
ICM-20948 + baro or the [ICM-20789](https://invensense.tdk.com/products/motion-tracking/7-axis/ICM-20789/) might be an appropriate alternative.
I thought about creating a board including a piezo buzzer to keep the space requirements small and figured out that the mpu6050 should not be used anymore. Doubt I'll have the time to work on the code, unfortunately.

Bernd

bzed, 2022-07-17:
https://www.st.com/en/mems-and-sensors/lsm9ds1.html is used by the skybean strato, might also be an option.

gereic, 2022-07-22:
Hi, and welcome,
I have looked at this bar-sensor `https://invensense.tdk.com/download-pdf/icp-20100-datasheet/`
Looks good, I think.
Can also do the programming ?

bzed, 2022-07-22:
> https://invensense.tdk.com/download-pdf/icp-20100-datasheet/

I think if you only want to replace the baro sensor - why not use the bme280 (didn't check if that will work on the same bus as the gyro stuff, though).

My idea was to replace the whole gy-86 board by a single chip, if possible.

bzed, 2022-07-22:
(Or by BME280 + LSM9DS1, for example. There is no current gyro + baro module on one board from adafruit, but I'd happily order some from jlcpcb and publish the project.

gereic, 2022-07-23:
I think the icp-20100 is a better BARO-Chip then the BME280.

bzed, 2022-08-04:
https://www.waveshare.com/product/modules/sensors/accelerometer-angular-rate-compass/10-dof-imu-sensor-d.htm
might be an option, even if its using the bme280 - but its a module that is just available for everybody to buy.
Or https://www.dfrobot.com/product-818.html which is a bit cheaper than the waveshare module.

The icp 20100 is not even aviable at jlcpcb/lscs, so creating a new module with it would be rather expensive.

bzed, 2022-08-17:
I've done some more investigation. The chips on the waveshare module (and the gy-912) are also retired. The current models like the icp20100: ordered, delivery unknown or next year... Same for the STM chip used by Skybean in the strato - retired, new models not available.

What seems to be available are chips from Bosch, BNO055 with a BMP388 might be an option.

closed 2025-02-19 by gereic