gereic/GXAirCom #109: Feature Request: Peet Bros anemometer support

mikechristiansen, 2022-10-24:
Hi, 
Is it possible to support a different anemometer? Peet bros make very reliable hardware. Less wear parts, bigger cups and vein than the Davis product. The issue is that this sensor uses 2 Hall effect switches to measure wind speed and direction. Speed is simply counting the speed on one of these switches but direction is calculated by the time differential between them. 
Code for this sensor has been worked out and validated well by this project. https://github.com/lkarsten/peetbros-arduino/blob/master/peetbros-arduino/peetbros-arduino.ino 
Not sure how to go about coding anything for this?
Just a thought if it could be integrated easily I would very much appreciate it.
Regards, 
Mike.

gereic, 2022-10-24:
Hi Mike,
As I have read, the Peet bros is running on 5V. The ESP32 is running on 3.3V
If there are only 2 reed-switches it can work also at 3.3V. 
Can you try that ?
I think it should be not a big deal to implement the aneometer.
The best thing would be, that you send me a anemeter, I will test it and send it back to you.

mikechristiansen, 2022-10-26:
Hey thanks so much for replying. 
Where are you based? Im in NZ so likely might cost quite a bit to send an anemometer to you. 
I have had one running on 3.3v using the code from the above adapted to my old home automation setup.
I do have spare T-Beam hardware here I could test on if that helps at all. 
Im willing to make a generous donation to the project to enable this anemometer.
Also would love to see compatibility for 869.259 FLARM in NZ. 
Thank you so much for all your hard work on this project! I can appreciate it is a lot of work! 
Regards, Mike.

gereic, 2022-11-01:
I am in Austria, so it would be better, I buy one in Europe.
Do you know, where I can get one in Europe?

mikechristiansen, 2022-11-01:
Im not sure on the best source in the Europe but I did find this.
https://www.meteo-shopping.com/en/sensors/400-130-anemometre-girouette-pro-12-metres-de-cable.html#/44-type_anemometre-pro
Thanks so much!

gereic, 2022-11-18:
Hi,
I ordered a peet-bros anemometer.
When I have it, I will implement the code and test it.

mikechristiansen, 2022-11-20:
Amazing thank you!

gereic, 2022-12-22:
Hi,
I got a message from UPS, that the peet-bros anemometer will arrive at december 28.
So once, it is arrived, I will implement it.

gereic, 2022-12-27:
Can you compile a new version yourself ?
If yes, you can try the latest code from master-branch
Peet Bros is implemented.
Please try it an give me a shout.

gereic, 2022-12-27:
In the readme.txt from the bin-directory, you can see, where you have to connect the anemometer.
You need 2 pull-up-resistors in order to get the anemometer working.

mikechristiansen, 2022-12-28:
Thanks so much. 
Not sure on how to compile, will look into this. 
Is it a big job to add compatibility for 869.259 FLARM for use in NZ? Hoping to install a few ground stations for use in my local area.
Thanks so much!

gereic, 2022-12-28:
The new version is online. V5.2.0
Just du an update and then you can choose the PeetBros from the list of anemometers in the settings.
Be sure that "operating-mode" is on "Ground-station" and you checked "ground-station is weather-station" in the settings, otherwise it is not visible.
I tried it with a T-Beam.
Here are some pictures for the wiring.
Please try it and give me feedback.

[image]
[image]

gereic, 2022-12-28:
> Thanks so much. Not sure on how to compile, will look into this. Is it a big job to add compatibility for 869.259 FLARM for use in NZ? Hoping to install a few ground stations for use in my local area. Thanks so much!

This will be the next step.
But only to be sure, with FLARM, you can't send the weather-data to the devices.
Weather-data are only sent by FANET.
Of course you can upload Weather-data to weather-underground or windy.
And you can track FLARM-Devices to OGN.

gereic, 2022-12-30:
Can you test the Flarm-part for NZ ?
I don't have a real Flarm-device, so can't test it.
If yes, I can send you a test-binary and instructions for it.

mikechristiansen, 2022-12-30:
Hi, 
Sorry I don't have a Flarm device. I can test receiving aircraft if that helps? I live next to an airfield so should get plenty of traffic.

gereic, 2023-01-02:
You can try the latest version V5.3.0
Best way is to take a T-Beam an set it to Air-Mode.
Switch on Flarm RX and Flarm TX only !! (in the settings)
Keep in Mind, that you need a GPS-Signal in order that Flarm-radio is working.
When all is ok so far, you can have a look in the info-page.
The FlarmTX-Counter should count up every second (when GPS has signal).
When a nearby aircraft is sending flarm (and my codechange is working), the Flarm-RX-Counter should count up.
If yess, you should see the aircraft on the "received aircraft" page.

Please tell me, if it is working.

closed 2023-07-16 by gereic