gereic/GXAirCom #118: Feature Request: Repeater feature

Xen01r, 2023-01-02:
Hello,

I would like to deploy a ground station on a hill next to my house where there is no internet connection and no electricity. If I could placed an autonomous (solar powered) repeater on this hill and a ground station at my home it would be perfect.

Would it be possible with the current project ?

gereic, 2023-01-03:
Hi,
You can do that already.

1. install a normal solar powered ground-station on the hill.
2. install a ground-station at your home.
3. adjust the settins at you home-ground-station, that it forward all weather-data to WU or WI.

thats all.

Xen01r, 2023-01-03:
Perfect ! Do you know what hardware should I use to minimize power consumption on the repeater side ?

I saw [here](https://github.com/gereic/GXAirCom/blob/44b949f5dbe84680bbc96384c22341b2b4be0393/_03parts/README.md) that some work has already be done on solar powered hardware. If I understood it well, it use a Lilygo TTGO T-Sim7000G Module which can connect to cellphone network via a SIM card and a Lilygo Lora 868Mhz Shield to connect to the LoRa network. That's not really what I wanted to do but it could also work !

gereic, 2023-01-05:
You can use a simple T-Beam and a solar-charger (TP4056).
Then you solder all wires directly to the board and it should work out of the box.
The wiring, you see in the readme.txt in bin-folder.

Xen01r, 2023-02-10:
I'm doing some measurements of current consumption right now on the T-Beam V1.1 with 18650 battery attached to it. I thought it could be useful for others people.
 
[image]
(a) Charging current of the 18650 battery (b) battery + T-beam V1.1 consumption.

Xen01r, 2023-02-14:
I try to optimize the current consumption of the T-beam V1.1 in ground station mode. I noticed that the CPU frequency plays a major role in consumption. By changing the default CPU frequency (240MHz) to 40MHz I save 30% of consumption (from 130mA to 90mA). Could you tell me what are the consequences of lowering the CPU frequency in ground station mode ?

closed 2023-11-03 by Xen01r