gereic/GXAirCom #121: Q: Does GXAirCom support Flarm V7?

rvt, 2023-02-04:
See Subject for the question :)

rvt, 2023-02-16:
I have to admit that I do not look into the source yet...

I am looking into stratux support for GXAirCom, for stratux that would possible mean:
1) Preferably faster GPS position reports (OGN supports 5 updates / sec)
2) To receive traffic reports via NMEA
3) Beable to set Aircraft ID for FLARM/OGN via nmea

So stratux users will beable to also be seen using GXAirCom, and use it at the same time as a GPS source. This is why we need to beable to set the aircraft ID from stratux because of it's management in the interface.

thoughts?

gereic, 2023-02-16:
Sorry,
Don't know what stratux is and what you want to do with GXAircom.
Can you explain more ?

rvt, 2023-02-18:
Hey,

the idea is that GXAirCom will communicate with Stratux just like it can do with OGN and SoftRF. Connection is with USB port and GXAirCom will send positional and time information aswell as traffic like PFLAA it sees around depending on the protocol. With OGN we also get number of satellites we see and more info.

This way we can see on our EFB (or web interface of stratux) the aircraft around us, but also we know we are seen by other aircraft because we transmit aswell.

closed 2023-04-24 by rvt