gereic/GXAirCom #125: Unable to open Web Server on T-Beam v1.1

Shropshire, 2023-04-15:
Hi, I've just bought a couple of T-Beam v1.1s and updated the software from GIT using Visual Code/PlatformIO. All looks good on the OLED..... BUT.. I'm unable to connect to the on board web server (192.168.4.1) - it seems to say Connection Rejected.

Thanks in advance,
Tony

rvt, 2023-06-02:
do you upload the filesystem? 

example:
`pio run --target uploadfs -e psRam`

maks, 2023-08-05:
@rvt  would you have some more details about the need to upload a filesystem? 

I tried to just follow the instructions in the wiki https://github.com/gereic/GXAirCom/wiki/Flashing-the-Software-with-a-PC-(Windows) though I'm on Linux and so used the flashall.sh shell script. It all seemed to work using the esptool and [my TTGO Tbeam v1.2](https://www.aliexpress.com/item/1005003088139358.html) seems to boot ok, though I do see some errors logged on the usb uart and I'm able to connect to the Wifi AP, but as soon as I try to the web server on 192.168.4.1 I see in the uart logging that it crashes and resets which seems same as the original report in this issue.

I flashed `firmware_v5.4.0_psRam.bin` with matching `spiffs_v5.4.0.bin` and the other binaries used `flash_all.bash` so I'm not sure what I'm doing wrong?

maks, 2023-08-05:
In case it helps here is the log, its for 5.3.4 since I tried it in case it was any better than the latest version but pretty much same behaviour:

```
I NimBLEDevice: NimBle host synced.
[I][ble.h:166] start_ble(): Starting BLE
[I][main.cpp:1422] WiFiEvent(): station start
[I][main.cpp:1503] setupWifi(): Setting soft-AP ... 
[I][ble.h:175] start_ble(): Waiting a client connection to notify...
[I][main.cpp:1506] setupWifi(): Ready
[I][main.cpp:1467] WiFiEvent(): soft-AP stop
[I][main.cpp:1464] WiFiEvent(): soft-AP start. IP: [192.168.4.1]
[I][main.cpp:1464] WiFiEvent(): soft-AP start. IP: [192.168.4.1]
[I][main.cpp:1511] setupWifi(): Setting soft-AP configuration ... 
[I][main.cpp:1513] setupWifi(): Ready
[I][main.cpp:1538] setupWifi(): hostname=GXAirCom-0803C0
[I][main.cpp:1425] WiFiEvent(): station stop
[I][main.cpp:1425] WiFiEvent(): station stop
[I][FanetLora.cpp:82] setGPS(): enable GPS PPS
[I][main.cpp:1470] WiFiEvent(): a station connected to ESP32 soft-AP
[I][main.cpp:1476] WiFiEvent(): soft-AP assign an IP to a connected station
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400d7aec  PS      : 0x00060f30  A0      : 0x800d7cf9  A1      : 0x3fff1660  
A2      : 0x3fff20b0  A3      : 0x00000000  A4      : 0x3fff16a4  A5      : 0x3ffcde18  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x800d7ae9  A9      : 0x3fff1640  
A10     : 0x3fff166c  A11     : 0x3f407905  A12     : 0x69cc703b  A13     : 0x3ffcde18  
A14     : 0x00000000  A15     : 0x3fff166c  SAR     : 0x00000017  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400956b1  LEND    : 0x400956c1  LCOUNT  : 0xfffffff9  

ELF file SHA256: 0000000000000000

Backtrace: 0x400d7aec:0x3fff1660 0x400d7cf6:0x3fff16a0 0x4011fa45:0x3fff16e0 0x4011d819:0x3fff1720 0x4011d8e1:0x3fff1760 0x4011db35:0x3fff17b0 0x4011ac79:0x3fff17d0 0x4011ad0d:0x3fff1810 0x4011b43d:0x3fff1830 0x40097bce:0x3fff1860

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
[I][FanetLora.cpp:84] setGPS(): disable GPS PPS
[I][main.cpp:1823] setup(): SDK-Version=v3.3.5-1-g85c43024c
[I][main.cpp:1824] setup(): CPU-Speed=240MHz
[I][main.cpp:1825] setup(): XTAL-Freq=40MHz
[I][main.cpp:1826] setup(): APB-Freq=80000000Hz
[I][main.cpp:1827] setup(): Total heap: 226780
[I][main.cpp:1828] setup(): Free heap: 199752
[I][main.cpp:1838] setup(): Total PSRAM: 4194252
[I][main.cpp:1839] setup(): Free PSRAM: 4194252
[I][main.cpp:743] printChipInfo(): This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, 
[I][main.cpp:744] printChipInfo(): silicon revision 3, 
[I][main.cpp:746] printChipInfo(): 4MB external flash
[I][main.cpp:1846] setup(): compiled at Jun 25 2023 17:58:13
[I][main.cpp:1847] setup(): current free heap: 4387612, minimum ever free heap: 4382780
[I][main.cpp:1715] print_wakeup_reason(): wakeup undefined --> possible by reset
[I][main.cpp:1857] setup(): SPIFFS total=113201 used=0 free=113201
[I][fileOps.cpp:6] load_configFile(): LOAD CONFIG FILE
[E][Preferences.cpp:492] getBytesLength(): nvs_get_blob len fail: BATOFFS NOT_FOUND
[I][main.cpp:1867] setup(): set CPU-Speed to 240MHz
[I][main.cpp:1602] printSettings(): **** SETTINGS v5.3.4 build:Jun 25 2023 17:58:13 ******
[I][main.cpp:1603] printSettings(): Access-point password=12345678
[I][main.cpp:1604] printSettings(): Board-Type=5
[I][main.cpp:1605] printSettings(): Display-Type=1
[I][main.cpp:1606] printSettings(): Display-Rotation=0
[I][main.cpp:1607] printSettings(): AXP192=0
[... 138 lines cut ...]
I NimBLEDevice: BLE Host Task Started
I NimBLEDevice: NimBle host synced.
[I][ble.h:166] start_ble(): Starting BLE
[I][main.cpp:1422] WiFiEvent(): station start
[I][main.cpp:1503] setupWifi(): Setting soft-AP ... 
[I][ble.h:175] start_ble(): Waiting a client connection to notify...
[I][main.cpp:1506] setupWifi(): Ready
[I][main.cpp:1467] WiFiEvent(): soft-AP stop
[I][main.cpp:1464] WiFiEvent(): soft-AP start. IP: [192.168.4.1]
[I][main.cpp:1464] WiFiEvent(): soft-AP start. IP: [192.168.4.1]
[I][main.cpp:1511] setupWifi(): Setting soft-AP configuration ... 
[I][main.cpp:1513] setupWifi(): Ready
[I][main.cpp:1538] setupWifi(): hostname=GXAirCom-0803C0
[I][main.cpp:1425] WiFiEvent(): station stop
.4.1
[I][main.cpp:1425] WiFiEvent(): station stop
[I][FanetLora.cpp:82] setGPS(): enable GPS PPS
[I][main.cpp:5500] taskBackGround(): ******************WEBCONFIG Setting - WIFI STOPPING*************************
[I][main.cpp:5501] taskBackGround(): currHeap:4252256,minHeap:4249860
[I][main.cpp:3713] setWifi(): switch WIFI OFF
[I][main.cpp:1467] WiFiEvent(): soft-AP stop
[I][main.cpp:1467] WiFiEvent(): soft-AP stop
[I][main.cpp:3731] setWifi(): set CPU-Speed to 240MHz

```

gereic, 2023-08-06:
It look like you haven't flashed the spiffs
[I][main.cpp:1857] setup(): SPIFFS total=113201 used=**0** free=113201

maks, 2023-08-06:
@gereic thank you for having a look at this and thanks for making such a great project!

I flashed using esptool on linux multiple time now with different versions of the binaries.
I tried again now on the latest published version with:
```
esptool -p /dev/ttyACM0 -b 921000 write_flash --flash_mode=dio 0x1000 bootloader_dio_40m.bin 0x8000 partitions.bin 0xe000 boot_app0.bin 0x10000 firmware_v5.4.0_psRam.bin 0x3d0000 spiffs_v5.4.0.bin
```

so I'm not sure with the spiffs why its not there? is `0x3d0000` the correct address for the filesystem? I just got that address from the flash_all bash script and its the same value as in the wiki.

I also just realised that I got my board from a vendor instead of from the "official" Lilygo store on Aliexpress, so I've ordered another unit from them just in case the vendor I bought from has sent me a dodgy unit.

maks, 2023-08-06:
sorry I should have also included the output from running esptool which seems to show the spiffs binary being written correctly to flash?

```
 ./flash_all.bash 
esptool.py v2.8
Serial port /dev/ttyACM0
Connecting....
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: b0:b2:1c:50:03:c0
Changing baud rate to 921000
Changed.
Enabling default SPI flash mode...
Configuring flash size...
Auto-detected Flash size: 4MB
Erasing flash...
Took 0.24s to erase flash block
Wrote 16384 bytes at 0x00001000 in 0.3 seconds (468.9 kbit/s)...
Hash of data verified.
Erasing flash...
Took 0.07s to erase flash block
Wrote 3072 bytes at 0x00008000 in 0.0 seconds (517.7 kbit/s)...
Hash of data verified.
Erasing flash...
Took 0.13s to erase flash block
Wrote 8192 bytes at 0x0000e000 in 0.1 seconds (517.3 kbit/s)...
Hash of data verified.
Erasing flash...
Took 6.78s to erase flash block
Wrote 1632256 bytes at 0x00010000 in 27.7 seconds (471.1 kbit/s)...
Hash of data verified.
Erasing flash...
Took 1.95s to erase flash block
Wrote 196608 bytes at 0x003d0000 in 3.1 seconds (509.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

maks, 2023-08-06:
hi again @gereic I just had a look a the code in main.cpp around line 1828 and it looks like you have changed the code recently and now have code that will format it on opening:
```
 // Make sure we can read the file system
  if( !SPIFFS.begin(true)){
  //if( !SPIFFS.begin(false)){
    log_e("Error mounting SPIFFS");
    //while(1);
  }
  log_i("SPIFFS total=%d used=%d free=%d",SPIFFS.totalBytes(),SPIFFS.usedBytes(),SPIFFS.totalBytes()-SPIFFS.usedBytes());
```
so I would expect that by calling `SPIFFS.begin(true)` with the bool set to true it will format the spiffs when its opened, so it would be expected to be empty.

Should I instead go back to using an older version of the code which still uses spiffs?

maks, 2023-08-07:
oh! I just saw you pushed up a commit with newly rebuilt 5.4.0 binaries, so I flashed with them and now I'm able to connect successfully to the http server! 
I'm not sure what specific change in one of the commits since the previous 5.4.0 bins fixed it, but very happy its working for me now on my board, thanks again @gereic !

closed 2025-02-19 by gereic