gereic/GXAirCom #128: Forward weather data from Weather Underground to FANET

MrMorice, 2023-05-15:
Dear Team, first a compliment to this great piece of software. A great contribution to the paragliding community!

With a couple of friends I would like to install a couple of weather stations in Wildschönau, which provide valley weather data via FANET. As we have some local weather stations in Weather Underground it would be great to forward their data via FANET as well.

My coding skills are a bit rusty and I can't even get Platform.io up and running at the moment, but happy to contribute a weather station once I finished my design.

Best, Mo

rvt, 2023-05-15:
If I am not mistaken, this is already in the code : https://github.com/gereic/GXAirCom/blob/master/src/main.cpp#L413

@gereic I did notice that this bit of code needs to be wrapped in a conditional compiler, when the compiler definition GSMODULE is not provided, compilation fails because `WeatherUnderground` is not compiled in. I think sendFanetWeatherData2WU/sendFanetWeatherData2WI needs to be wrapped in `#if defined(GSMODULE)` and the calling functions at line https://github.com/gereic/GXAirCom/blob/master/src/main.cpp#L4648 and https://github.com/gereic/GXAirCom/blob/master/src/main.cpp#L4654

I can provide a PR..

gereic, 2023-05-15:
You can already send 1 WU-Station by 1 GxAircom to FANET.
Sending multiple WU-Stations from 1 GxAircom is not implemented yet.

MrMorice, 2023-05-15:
Thank you both for the super quick response 👌!

@rvt One station is already a good start, multiple would be amazing, as we have five good WU stations in the valley 🎉.
@gereic Cant seem to find the settings in 5.3.3.

closed 2025-02-19 by gereic