gereic/GXAirCom #130: Build from source documentation ?

jccricket, 2023-06-28:
Hello,

I have AXP2101 and I would like to adapt code (in the board detection part - and other places where it code should be changed).
I was wondering if there are some existing documentation about "build from source" so I can play around.

Thank you.

(the issue I'm facing is mainly related to power management (battery icon being empty, apparently not charging battery when plugged in. I'm using a T-Beam v1.1 that has sx1276 but detected as 1262 by the GXAirCcom code - as per logs - and this sx1276 has AXP2101 v1.2 where apparently most of sx1276 has AXP192. Mine looks recent: 20230508 is written on the board next to the antenna)

bzed, 2023-06-28:
sx1276 is a lora transciever, AXP192 and others are power management related. 
sx1276 is basically an sx1262 chip with some additional functions.

So I can't see an issue there...

The project is using platformio, I would suggest visual studio code as editor, but everything else should just work well, too.

jccricket, 2023-06-28:
I am super new into this, but when ready this (https://github.com/gereic/GXAirCom/blob/master/lib/AXP202X_Library/src/axp20x.h):
#define AXP202_SLAVE_ADDRESS (0x35)
#define AXP192_SLAVE_ADDRESS (0x34)

I have the feeling they do not share the same...

jccricket, 2023-06-28:
Just to come back to the initial question about how to build from source documentation:

I have managed to achieve my goal: copy_file([".pio/build/psRam/firmware.bin"], [".pio/build/psRam/firmware.elf"])
Copying file output to project directory...
firmware.bin
psRam
build
v5.3.5
********  copy file .pio/build/psRam/firmware.bin to bin/firmware_v5.3.5_psRam.bin *******
Done.
=============================================================================== [SUCCESS] Took 30.20 seconds ===============================================================================

Environment    Status    Duration
-------------  --------  ------------
psRam          SUCCESS   00:00:30.203

Here are the steps I have followed for those who are interested:
1. install visual studio code ($sudo apt snap code)
2. install platformio plugin (https://docs.platformio.org/en/latest/integration/ide/vscode.html#ide-vscode)
3. fork this github project into my github space
4. create a branch
5. open a git project from visual studio code (in platformio plugin in visual studio code, there is a quick access drop down, in which I've found Miscellaneous/Clone Git PRoject)
6. move to the branch I created
7. went to project task, and run the build for psRam profile
8 done, compiled with many (29) warning, but compiled

closed 2023-06-28 by jccricket

jccricket, 2023-07-04:
Just adding this here for reference for those who will come and see:
https://github.com/lewisxhe/XPowersLib

mooiweertje, 2024-02-10:
Hi, Is there a downloadable firmware for the 1.2 board? The ones in the repository all don;t work on my board.