gereic/GXAirCom #132: PlatformIO build issues - WiFi access resets T-Beam

VirusPilot, 2023-06-29:
I just successfully built and flashed the binaries, all works fine but only when connecting the GxAirCom web page via WiFi it resets the T-Beam. This is not the case when flashing the upstream binaries.

FYI, I have compiled/flashed the psRAM version for my T22_V1.1 board.

jccricket, 2023-06-30:
I faced the same issue yesterday. (building with PlatformIO)
Thanks for the info.

On my board, I have on the right side a sticker with MODEL:T-BEAM-v1.1
Close to the antenna, I can read TBeam-AXP2101-V1.2

(I'm mentioning about this AXP2101, as I wanted to be able to build code myself to better understand how power management works. So far, I am able to build, but debug doesn't work. A separate issue will be opened when I'll progress more - lack of time at the moment)

VirusPilot, 2023-07-03:
I am not having this issue any more, it might really be the case that this issue only arises in case no prior version was on the device (and with no saved settings). So I will close the issue.

closed 2023-07-03 by VirusPilot